freifunk-gluon/gluon #259: Refactor gluon-announced and gluon-neighbour-info

tcatm, 2015-01-05:
Both gluon-announced and gluon-neighbour-info should be refactored into a toolsuite for general multicast information distribution. All dependencies on gluon itself should be removed.

NeoRaider, 2015-07-12:
Is anything about gluon-neighbour-info specific to gluon besides its name?

tcatm, 2015-07-12:
No. It's just a client for respondd (with some special feature like HTTP output when used as a CGI script).

NeoRaider, 2015-10-11:
So... what would be a nice name for gluon-neighbour-info? "multicast-query"?

tcatm, 2015-11-02:
I'd like to keep the name gluon-neighbour-info at least for 2015.2 as for example the SSE feature is tailored to use with the status page.

Beyond 2015.2 I could imagine gluon-neighbour-info to morph into a more general "gluon-node-information" tool that can collect information both from the local node as well as from neighbours or even nodes that are further away. I don't see it remaining a simple multicast tool for a long time.

NeoRaider, 2015-11-02:
Okay, so let's close this for now.

closed 2015-11-02 by NeoRaider