freifunk-gluon/gluon #260: Clean up dependencies

tcatm, 2015-01-06:
I've noticed that dependencies are in a messy state. For example, gluon-core does not depend on lua directly while it's obviously required for proper operation. Therefore I suggest reviewing all gluon packages for such indirect dependencies and adding them to the Makefiles.

NeoRaider, 2015-07-12:
Not specifying dependencies which are obviously satisfied indirectly is common practise (gluon-core depends on luci-base, which obviously pulls in lua).

.The luci dependency situation has become worse in Chaos Calmer: 
* luci-base now depends rpcd, even though it wouldn't be required for Gluon, as rpcd is used for authentication only. This not only wastes flash, but also RAM, as rpcd is actually running (it's pretty small though)
* luci.util won't work at all unless ubus-lua is installed (this dependency of luci-base is missing though)

I guess we might be at a point where forking LuCI to create a minimal version would make sense.

NeoRaider, 2016-02-28:
I've added a script `contrib/depdot.sh` which will output the dependencies of all Gluon packages as a dot graph.

I think relying on obvious dependencies is okay (for example LuCI pulling in Lua, or all Gluon packages somehow depending on gluon-core and thus gluon-site), but we should look for non-obvious dependencies and make them explicit to avoid accidential breakage.

NeoRaider, 2016-07-27:
I've studied our dependencies a bit and didn't find obvious issues. Also, `luci-lib-nixio` and `luci-lib-jsonc` don't pull in luci-base anymore.

I think this can be closed; if we find an unobvious dependency, we'll just fix it.

closed 2016-07-27 by NeoRaider