freifunk-gluon/gluon #263: Unterschied Namensvalidierung Config Mode / Router Registrierung

tim-bec, 2015-01-07:
Vergibt man im Config Mode einen Namen mit einem Punkt ( z.b. "ffac-mein.knoten")  wird dies akzeptiert, das registrierungsformular ( http://register.rheinufer.freifunk-rheinland.net/ ) lehnt aber namen mit einem Punkt ab.

CodeFetch, 2015-01-07:
Das hat nichts mit Gluon zu tun. Die Webseite ist kein Standartformular. Du hast vor 15 Minuten irgendwas im Freifunk Rheinland Forum geschrieben. Dann frag doch da, bevor du hier bei den Entwicklern der Firmware anfragst.

tim-bec, 2015-01-07:
Dann wäre es doch prima, wenn Du mir einfach den Link zum entsprechenden git gibst anstatt dich auf irgendwelche Posts von mir im ff forum zu beziehen...

NeoRaider, 2015-01-07:
Während das Formular nichts mit Gluon zu tun hat, ist es in der Tat komisch, dass der Config Mode Punkte erlaubt. Punkte sind aber auch nicht das einzige Zeichen, das wir da mal ausschließen sollten.

Auf Dauer ist der Plan, einen "Pretty Name" einstellbar zu machen, der einfach jedes Zeichen erlaubt (wie das auch die meisten Desktop-Betriebssysteme haben), aus dem dann der Hostname generiert wird.

mrs4ndm4n, 2015-01-07:
Warum so unfreundlich?

tim-bec, 2015-01-07:
https://github.com/freifunk-gluon/packages/blob/master/gluon/gluon-config-mode-hostname/files/lib/gluon/config-mode/wizard/0100-hostname.lua scheint die Stelle zu sein. Da wird nichts validiert, stellt sich jetzt nur die Frage ob die Validierung in das einzelne Package gehört oder globaler...

tcatm, 2015-01-07:
@tim-bec: Das `o.datatype = hostname` weist luci an das Feld als "hostname" zu validieren. Der Code dazu ist hier zu finden: https://github.com/openwrt/luci/blob/master/modules/base/luasrc/cbi/datatypes.lua#L180

tim-bec, 2015-01-07:
@tcatm cool! und mit wem müssen wir nun alles sprechen das das angeglichen wird, bzw. wer stellt den pr? ;-)

CodeFetch, 2015-01-07:
Man wird den Hostnamen immer ändern können. Die Gluon-Firmware ist ja auch nur OpenWRT. Deshalb habe ich ehrlich gesagt noch nicht das Problem mit den Punkten verstanden... Soweit ich weiß sind die Hostnamen doch nur für das Mesh-Netz relevant?! 

Frag doch bitte im Freifunk Rheinland Forum. Nicht jede Community hat dieses Formular. Ich kenne das git nicht wo dieses Formular drin ist, wenn es denn eines gibt. 

Das Formular funktioniert sowieso nicht richtig. Man kann auch bei Gluon wesentlich längere Knotennamen eintragen als das Formular akzeptiert. 
Außerdem kann es passieren, dass ein Knoten sich nicht mehr mit mit dem Gateway über fastd verbinden kann, wenn der Key mehrmals eingetragen wurde. 
D.h. du hast einen Knoten eingerichtet und als Queen benutzt und dann bringst du ihn in den Config mode, änderst den Namen, trägst den Key mit neuem Namen zum zweiten Mal in dieses Formular ein und auf einmal kann der Knoten sich nicht mehr über fastd verbinden. 
Wenn du hingegen nur den Knotennamen z.B. über UCI änderst, läuft er fröhlich weiter. 

Also ich wollte nicht unhöflich erscheinen, wenn das so empfunden wurde?! Mir geht nur dieses Formular selbst schon seit einiger Zeit auf die Nerven und ich würde mich freuen, wenn da mal jemand dem Zuständigen diese Fehler mitteilt.

tcatm, 2015-01-07:
@tim-bec Möchtest du? Es wäre wohl sinnvoll einen weiteren Validierungstyp einzufügen, da `hostname` offenbar auch FQDNs abdecken soll. Sinnvoll wäre wohl eine Beschränkung auf a...z, A...Z, 0...9 und den Bindestrich. Das sind alle Zeichen, die in einem DNS Eintrag vorkommen dürfen. Einige sind schon darüber gestolpert, dass Unterstriche zwar angenommen wurden aber dann im DNS Server der Community nicht eingetragen werden konnten.

tim-bec, 2015-01-07:
@CodeFetch Schon vergessen und sorry für die harsche Antwort ;-) Ändern des Namens ist ja nicht das Problem, sondern das der Config Mode einen Punkt erlaubt, das Formular ( https://github.com/ffrl/ffff/blob/master/static/index.html ) dann aber sagt: "Hey, Punkt ist nicht." - Ist halt so rum doof, da man dann wieder ins config muss, den namen ändern, noch mals registrieren. 

Die Idee von @NeoRaider mit den Pretty Names scheint da wohl die klügste zu sein. 

@tcatm  - Problem aber an dem PR ist wohl das die Validierung in openwrt stattfindet und nicht im gluon Kosmos... Ich könnte den PR erstellen, aber nicht wirklich prüfen da mir dazu dann doch das Entwickler Handwerk fehlt.

tcatm, 2015-01-07:
@tim-bec Ja, der PR müsste am besten gegen luci erstellt werden. Wir können den dann als Patch mit in Gluon aufnehmen bevor er dort angenommen wird. 

In http://gluon.readthedocs.org/en/v2014.4/dev/basics.html ist grob beschrieben, wie man in Gluon Patches einfügt. Im Prinzip hängst du einen Commit im Teilbaum `packages/luci` an den Branch `patched` an (ganz normales `git commit`), testest das Image und wenn alles läuft, machst du im Hauptverzeichnis von Gluon `make update-patches`. Das erzeugt aus deinem Commit einen Patch in `patches/`, den du uns dann zukommen lassen kannst.

CodeFetch, 2015-01-07:
[-a-zA-Z0-9_]{1,32} in https://github.com/ffrl/ffff/blob/master/server.js ist wohl das Problem.
Wie wäre es denn stattdessen mit [-a-zA-Z0-9\.]{1,64}?

Also ich verstehe ehrlich gesagt noch nicht, wieso man keine Punkte erlauben sollte. Ergänzt der DNS-Server denn nicht sowieso immer z.B. ".ffrg" falls er überhaupt Knoten automatisch bei sich einträgt (welche Community macht denn überhaupt sowas)? In den Fällen wo man Hostnamen spoofen kann, wird das auch so bleiben (z.B. mehrere Clients an einem Router untereinander).

hhansen06, 2015-01-07:
Warum nicht RFC 952 konform? Da haben sich schonmal leute gedanken drüber
gemacht :)
Das hat beim DoD funktioniert, warum nicht auch bei Freifunk?
https://tools.ietf.org/html/rfc952

1. A "name" (Net, Host, Gateway, or Domain name) is a text string up
   to 24 characters drawn from the alphabet (A-Z), digits (0-9), minus
   sign (-), and period (.).  Note that periods are only allowed when
   they serve to delimit components of "domain style names". (See
   RFC-921 <https://tools.ietf.org/html/rfc921>, "Domain Name System
Implementation Schedule", for
   background).  No blank or space characters are permitted as part of a
   name. No distinction is made between upper and lower case.  The first
   character must be an alpha character.  The last character must not be
   a minus sign or period.  A host which serves as a GATEWAY should have
   "-GATEWAY" or "-GW" as part of its name.  Hosts which do not serve as
   Internet gateways should not use "-GATEWAY" and "-GW" as part of
   their names. A host which is a TAC should have "-TAC" as the last
   part of its host name, if it is a DoD host.  Single character names
   or nicknames are not allowed.

CodeFetch, 2015-01-07:
Für mich hört sich das an wie IPv4 :D... Derzeit werden viele Knoten in dem Format FF-[KFZKennzeichen oder Stadtname]-[Stadt oder Stadtteil]-[Nick oder Straßenname oder Firmenname o.Ä.] bezeichnet. Das wäre dann nicht mehr möglich. DNS sollte nach rfc2181 oder so auch mit Sonderzeichen klarkommen. Ein Domainname darf danach 255 bytes lang sein.

tcatm, 2015-02-22:
@tim-bec Wie sieht's aus? Gibt's den PR mitlerweile? :)

NeoRaider, 2015-07-12:
This is now tracked in #414.

closed 2015-07-12 by NeoRaider