freifunk-gluon/gluon #264: Management Interface NanoStation mit auf br-client

hhansen06, 2015-01-07:
Um das Management einer an einen Gluon Knoten angeschlossenen NanoStation zu erleichtern, haben wir die Möglichkeit getestet auf dem WAN Anschluss (wr841n V8 = eth0) ein 8021q Vlan welches auf br-client gebrückt ist einzurichten.

Die installation bei einem potentiellen Freifunker wird deutlich überschaubarer, die Nanostation bleibt konfigurierbar, es lässt sich AirControl einsetzen ohne das man ein seperates Subnetz im eigentlichen batman-adv Traffic mitschleifen muss.

Hierfür müsste folgendes in die /etc/config/network rein:
```
config interface 'client'
        ...
        option ifname 'eth1 bat0 eth0.10'
       ...
```

Schön wäre es wenn es im Modul mesh-on-wan zusätzlich noch eine Checkbox in den Expertenmodus eingebaut werden könnte.

Zusätzlich sollte dort dann noch der Warnhinweis angebracht werden, des es vermieden werden sollte mehrere gluon Knoten wan an wan mit diesem "feature" zu konfigurieren.

tcatm, 2015-01-07:
Um solche "Spezialkonfigurationen" abzufangen haben wir https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration eingerichtet. Vielleicht könntest du dort die nötigen uci-Befehle (d.h. keine Anleitung wie man Dateien editiert) hinterlegen. Pluspunkte gibt's für die Verwendung von `/lib/gluon/core/sysconfig/*`. Beachte, dass es sowohl Geräte mit `eth0` und `eth1`, als auch welche mit nur einem VLAN-Switch gibt.

hhansen06, 2015-01-07:
auch ne idee. die uci configs bleiben nach einem update erhalten?

tcatm, 2015-01-07:
Nicht zwingend. Aktuell ist nicht dokumentiert, was erhalten bleibt, aber vieles überlebt tatsächlich. Am besten liest du mal die invariant upgrade Skripte der einzelnen Gluonpakete.

hhansen06, 2015-01-07:
Habs mal in die Liste mit aufgenommen. 
https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#nanostation-webinterface-im-freifunk-netz-

Kann dann geschlossen werden.

tcatm, 2015-01-07:
Cool. Bedenke, dass auf einigen Geräten wan_ifname auch sowas wie "eth0.2" sein kann. Da einfach .10 (eth0.2.10) dranzuhängen, funktioniert wahrscheinlich nicht.

hhansen06, 2015-01-07:
hm ... wenn ich die openwrt doku richtig verstehe, geht 8021q eh nur an interfaces wo kein vswitch hinter hängt. 
ich schreib's mit in den text. wir verwenden aktuell nur 841N V7 und 8 da geht das so :D

closed 2015-03-03 by tcatm