freifunk-gluon/gluon #266: MTU 1

hhansen06, 2015-01-08:
Wenn ich das open-mesh Wiki richtig verstehe, sollte doch die MTU auf den Interfaces die batman-adv transportieren auf 1528 gesetzt werden oder?

http://www.open-mesh.org/projects/batman-adv/wiki/Fragmentation-technical

tcatm, 2015-01-08:
Wo ist das Issue? Allgemeine Fragen bitte über die Mailingliste oder im IRC.

closed 2015-01-08 by tcatm

hhansen06, 2015-01-08:
Okay dann anders...
 
Teststellung:                                             
[10.213.2.10] < - - > [gluon_a] < - batman_adv - > [bluon_b] < - - > 10.213.0.10

auf 10.213.0.10: 
# ping 10.213.2.10 -s 1400
ergiebt per batctl td auf gluon_a:
19:41:08.016533 IP 10.213.0.2 > 10.213.2.10: ICMP echo request, id 3913, seq 1, length 1429
19:41:08.017291 IP 10.213.2.10 > 10.213.0.2: ICMP echo reply, id 3913, seq 1, length 1429

# ping 10.213.2.10 -s 1500
ergiebt per batctl td auf gluon_a:
19:41:31.987299 IP 10.213.2.10.59402 > 172.16.0.2.9081: TCP, flags [.S....], length 0
19:41:33.084831 IP 10.213.0.2 > 10.213.2.10: ICMP echo request, id 3914, seq 1, length 1480
19:41:33.088227 IP 10.213.0.2 > 10.213.2.10: ICMP type 192, length 28
19:41:33.088707 IP 10.213.2.10 > 10.213.0.2: ICMP type 216, length 22

Wenn ich also den debug output richtig verstehe, wird aktuell fragmentation erzwungen?

tcatm, 2015-01-08:
Da fragmentiert gerade dein normaler Netzwerkstack, da auf `-s` beim Ping nochmal Header für ICMP und IP drauf kommen. Sagt dir ping auch: 1500(1528) bytes of data.

Fragmentierung ist auch kein Problem und ganz normal. Gluon ist trotzdem so eingestellt, dass es, wo immer möglich, vermieden wird. Insbesondere um doppelte Fragmentierung zu vermeiden.

FreifunkUFO, 2015-01-23:
siehe auch: https://github.com/freifunk-gluon/gluon/issues/275