freifunk-gluon/gluon #271: Multicast Firewall blocks server-free Serval Mesh VIOP Client discovery on UDP port 4110 (solution is here)

Moobooo, 2015-01-14:
in

"/lib/gluon/ebtables/"     (110-mcast-allow-serval)

a rule is missing that is needed to search for new serval voip devices

rule 'MULTICAST_OUT -p IPv4 --ip-protocol udp --ip-destination-port 4110 -j RETURN'

Please insert this rule in the next versions of gluon to enable encrypted, server free voip call and text messaging functions by the serval mesh for android app.

wireshark cap:
from:10.53.78.212	to:10.53.255.255 Source port: g2tag  Destination port: g2tag 

http://www.servalproject.org/
https://forum.freifunk.net/t/telefonieren-mit-serval-fur-android-und-probleme-mit-der-gluon-broadcast-firewall/

NeoRaider, 2015-01-14:
Hmm, I guess it would make sense to make `gluon-ebtables-filter-multicast` more configurable. At the moment we block almost everything as batman-adv doesn't scale well to big numbers of nodes and clients because of its huge broadcast domain (and a community that doesn't want to patch the filter package has just two choices: allow all multicast traffic, or block all with very few exceptions).

We are aware that it is very unfortunate to block so many useful decentralized services, but we had quite a problem with the broadcast load originating from Avahi and similar services.

T-X, 2015-01-19:
The idea of decentralized service discovery (and with that phoning/texting/...) is awesome and would really fit to our networks. Unfortunately, there doesn't seem to be an out-of-the-box, scalable solution available yet. That's why we unfortunately had to filter mDNS/DNSSD, too. Sending a broadcast packet every five seconds as the Serval does is way too much and won't work for us.

@Moobooo: @mwarning had been working on a DHT-based service discovery, KadNode (https://github.com/mwarning/kadnode). I'm still curious about seeing an mDNS-to-KadNode proxy application. That'd enable a whole range of decentralized applications out-of-the-box in a scalable fassion.

T-X, 2016-10-19:
Should we close this ticket until some protocol improvements to the overhead of the discovery protocol of Serval is available?

Paul Gardner-Stephen, developer of/at Serval, seems to be an approachable guy, he might be open to enhancements, I'd guess.

christf, 2016-12-03:
@T-X @NeoRaider  is it an option in the meantime to block these packets when they would be sent out on mesh-vpn so this stuff works locally while not loading the mesh. After all, this is the scenario when ad-hoc-services are actually useful.

jplitza, 2016-12-03:
@christf Regrettably (or maybe that's a good thing), there is no easy way to filter batman traffic. We can filter multicast traffic while it passes from the bridge to bat0, but we cannot filter it after that based on which mesh interface it uses (VPN, Wifi, LAN).

NeoRaider, 2020-05-10:
Closing in favour of #2020.

closed 2020-05-10 by NeoRaider

T-X, 2022-01-18:
@Moobooo I installed and tried the "Serval Mesh" app from the Fdroid store. I see these broadcast packets in tcpdump, too (with g2tag being port 4110). But I only see it for IPv4 and not for IPv6. If Serval Mesh were adding IPv6 multicast support then it should in theory work now at least with a limited number of "Serval Mesh" clients (< 16) if the network is updated to a recent Gluon version ([Gluon v2021.1](https://gluon.readthedocs.io/en/v2021.1.x/releases/v2021.1.html#multicast-optimizations-batman-adv)).

If you'd file a feature request ticket upstream (maybe [here](https://github.com/servalproject/batphone/issues)?) I'd be happy to provide guidance on how to add IPv6 multicast support to the "Serval Mesh" app. (But not sure if "Serval Mesh" is actually still developed or maintained, it didn't have any activity in the last three years, it seems).