freifunk-gluon/gluon #272: Add support for Ubiquitis UniFi AP Outdoor+

thisco-de, 2015-01-14:
As the *UniFi AP Outdoor+* is the successor of of Ubiquiti's *UniFi AP Outdoor* (non-plus version) it becomes more and more important to support this device with gluon, as it is hard already to obtain the non-plus version from any reseller.

There exists a [patch](https://lists.openwrt.org/pipermail/openwrt-devel/2014-December/030281.html) by @NeoRaider based on [this patch](https://lists.openwrt.org/pipermail/openwrt-devel/2014-June/026087.html) (?) for the current OpenWRT development branch (*Chaos Calmer*) that has hardly been tested by now and is not yet integrated into Gluon.

NeoRaider, 2015-01-14:
I'd like to get at least a little amount of feedback for the patch before integrating it into Gluon, especially because some people on openwrt-devel have seen WLAN issues.

thisco-de, 2015-01-14:
I created an integration patch for Gluon today, but wasn't able to test it on an actual device, yet. Our community (Paderborn) will have two of these devices by next week, so feedback is on its way ...

MPW1412, 2015-03-05:
Any updates on this?

NeoRaider, 2015-03-07:
Sorry, no good news. I've added the Gluon support as BROKEN as the WLAN doesn't work correctly. The signal is strong, but there's extremly high packet loss. I'm not sure what needs to be done to fix this. It might be possible to analyze the stock firmware...

MPW1412, 2015-03-07:
Thanks for the update. Can you do that? Do you need a testdevice? Or isn't that in your skill area?

NeoRaider, 2015-03-07:
I have a test device here. I'll try to find out more about the issue when I have some spare time, but it may be beyond my abilities. I know a few things about the internals of ar71xx hardware, but WLAN hardware and drivers are very complex.

oakey-dev, 2015-04-07:
Only so that it doesn't get forgotten.
I currently have to Bugs besides the WIFI Bugs from the openwrt-devel:
1. the MAC doesn't match the MAC from the Label
2. br-wan is bridged to the "secondary" Interface

mmalte, 2015-09-11:
Is there an update on this device? I'll get some of these within the next few weeks.

NeoRaider, 2015-10-26:
Unfortunately, the module to configure the HSR filter doesn't seem to work correctly, reception is still very bad.

commit 7c67ff0c (2016-01-17): ath9k: fix HSR tuner support for UniFi Outdoor Plus

Also remove the BROKEN check as the UniFi Outdoor Plus support seems to
work fine now.

Fixes #272
closed 2016-01-17 by NeoRaider