freifunk-gluon/gluon #277: IPv6 gateway selection

NeoRaider, 2015-01-19:
We must implement IPv6 gateway selection as currently
* The IPv6 gateway is selected randomly
* When there are too many router advertisers, nodes and clients easily get overwhelmed with handling all RAs

What needs to be done on the nodes:
* Make a list of available routers (we can probably just generate the list from received RAs... I think forking odhcp6c for this might make sense)
* Choose the best node (similar to batman-adv's gateway feature)
* Limit RSs and RAs using ebtables

tcatm, 2015-01-19:
Couldn't we make the node's anycast v6 the default gateway for nodes? That way the best gateway can even change when roaming without much trouble.

If we could disable RA handling on odhcp6c completely, there would be no need to fork. We could then write a new daemon that listens for RAs, forwards prefixes to the clients and sets the best default gateway locally.

tcatm, 2015-02-22:
I've did some experiments with bird6 and the node being an anycast gateway.

- bird6 accepts routes through RIP
- run radvd on local-node interface

It worked pretty well, does not require fiddling with the RAs and provides fast failover in case of disappearing gateways.

jplitza, 2015-08-03:
Using the node with its anycast address as gateway also has the advantage that the router announcements by the original gateways can be aggregated when passed to the client.

It also enables one special scenario: If multiple gateways can handle the same prefix (f.e. the /48 the community owns), it might make sense if each of them announces its own /64 (to the mesh and via BGP to the outside world, the latter in addition to the larger /48) so routes are symmetric. But it can also handle outgoing traffic from another gateways /64 thanks to the shared larger prefix.  If we have a software on the node that interprets this setting, we can take advantage of it. But this is totally optional.

However, having bird6 on the node seems like overkill: It's over 200 KB in size and handles much more than we need. Also, I'd advocate for using (slightly modified) RAs for announcing possible gateways, as this is downgrades gracefully on older nodes and clients connected directly, not using a Gluon node. The modification of the format could use a reserved field (f.e. the Reserved2 in the [Prefix Information Option](https://tools.ietf.org/html/rfc4861#section-4.6.2)) to specify the larger prefix for which a gateway is able to handle traffic.

To achieve all of this, we would have to:

1. Interpret incoming RAs
2. Insert appropriate default routes into the kernel (with a `from` that matches the advertised prefix and a metric calculated from the link quality to the router), for *all* RAs we receive
3. Send RAs with some of the prefixes (based on metric) from local-node to clients (this is partially already done by gluon-radvd, which could probably be modified to get the prefix information via netlink)
4. Optionally: Announce own IPv6 prefix on br-wan into the mesh if the user wishes to share his IPv6 uplink

I think the most difficult part would be 2.

paulinsche, 2016-09-09:
In the documentation of gluon-ebtables-filter-multicast I read: „The following packet types are considered essential and aren’t filtered: ICMPv6“

I would suggest, to just filter RAs at the gateways. This will solve the problems if a mesh is connected just through on server (with a second as backup). If a mesh is connected through two or more gateways, the solutions discussed here, will solve the problem in upstream, but not necessary in downstream.

RalfJung, 2016-09-09:
I have yet to find a way to filter the traffic Batman forwards through a Gateway. It doesn't pass through ebtables nor iptables...

jplitza, 2016-09-09:
What @RalfJung said: Filtering traffic on the gateways isn't easily possible, for example because of fragmentation that batman-adv does internally. You would have to implement an in-kernel filtering interface in batman-adv to do such filtering easily and reliably.

NeoRaider, 2016-09-09:
I agree that the gateway selection should happen on the nodes.

"The following packet types are considered essential and aren’t filtered: ICMPv6" means we can't just filter *all* ICMPv6 multicast packets, but we already filter some subtypes.

rotanid, 2018-01-19:
@jplitza is your PR #838 fixing this issue or am i on the wrong track?

jplitza, 2018-01-19:
@rotanid Correct.

closed 2018-11-14 by mweinelt