freifunk-gluon/gluon #287: atk9k: Stale wifi on TP-Link 1043nd v1

T-X, 2015-01-27:
About once in a week the wifi stales on an 1043nd here. With some effort I can get some replies via ping6 and even send some commands via ssh, but that window seems to be for about five seconds every three minutes or so. Seems similar to the issue we had a while back in one of the earlier releases?

Did a not quite stable feature get reenabled in barrier breaker? Since mostly we have a lot less packet loss on long / non-optimal wifi links (a lot more and greener links in the graph) but the issue of stale wifi again, maybe it has something to do with Adaptive Noise Immunity (ANI)? Did something like that get reenabled in barrier breaker?

Various logs from the 1043 can be found here: http://metameute.de/~tux/Freifunk/wifi-stats/

tcatm, 2015-01-27:
1043nd v1 and 941nd v1 (same wifi IIRC) at &nbsp; are having issues, too. Even on Chaos Calmer. Maybe it's best to just avoid ar91xx based devices.

CodeFetch, 2015-01-28:
I had the same issues with mine, but since it became colder outside (it sticks on a window) it's the most reliable Freifunkrouter I've ever owned.
Try to lower the TX power. That might be the problem.

tcatm, 2015-03-03:
We've decided to not recommend using ar91xx based devices.

closed 2015-03-03 by tcatm