freifunk-gluon/gluon #289: enable STP on br-client interface

flokli, 2015-02-06:
Sometimes (especially if you want to use already deployed wifi hardware) you want to have some kind of redundancy when bringing Freifunk connectivity.
Thats why one could decide to connect cables from different nodes together into the same layer2 network.

To prevent packet storms, you can use STP, which ensures that there is always only one path from one device to another (which would adapt after some time if one router goes down)

When both nodes are connected via different ports of a switch with STP already enabled, it works out of the box.
However, enabling on STP on the bridge interface of each Freifunk Node should't hurt either, but instead allow everybody to painlessly bridge the lan ports of two routers together directly.

NeoRaider, 2015-02-06:
Nope, this wil break. As STP packets are filtered by batman-adv, STP won't recognize the loop (Bridge A) -- (Bridge B) -- (bat0 on B) --mesh-- (bat0 on A) -- (Bridge A).

We've thought about enabling the feature "Bridge Loop Avoidance" (BLA) of batman-adv before, which is supposed to recognize such loops and disable forwarding. But as far as we could tell, BLA wasn't really as reliable as we'd liked. So far our choice was to disable it, so creating a loop will break things always and visibly instead of occasionally or sublty.

flokli, 2015-02-06:
Hmm, I think we should try to to work as good as possible, instead of willingly break some in some cases that could be fixed otherwise...
As STP traffic doesn't get forwarded by batman-adv, it shouldn't produce any more traffic in the global network, but fixes some situations we can fix easily.
When did you checked BLA last time?

tcatm, 2015-02-06:
STP sounds interesting for a segment connected to the LAN ports of multiple nodes. Maybe you could try it and document your findings (including internal switch state) with 3 or 4 devices?

NeoRaider, 2015-02-06:
As I said, using STP without BLA will definitely create a forwarding loop. It's been a long time since we've tried BLA, but I think it was already the new BLA2 version.

In any case, BLA is not perfect, and connecting two nodes' LAN ports will create a forwarding loop at least before BLA has detected the loop. This may be acceptable in a mesh with only 10-20 nodes, but completely insane in our Freifunk networks. So in my opinion the only conclusion can be: Don't do that. Ever. At all. Don't even think about it.

closed 2015-03-02 by tcatm