freifunk-gluon/gluon #290: Odd behavior of Wifi Switch TL-WDR3600

mmalte, 2015-02-06:
There is a "Wireless on/off" switch at the Back of the WDR3600

A fresh gluon 2014.4 install with the switch in the "on" position works fine.

Turning it to "off" results in a short wifi outage, possibly a /sbin/wifi, but the wifi is running fine afterwards

Turning it from "off" to "on" shuts down the wifi permanently.

Powercycle with switch in "on" position has no effect, wifi still is not operational.

Turning it to "off" results in the wifi working like a charm.

I was able to reproduce this behavior with three devices running the Rheinufer Image:
http://images.freifunk-rheinland.net/images/gluon/rheinufer/2014.4-stable-2/
https://github.com/ffrl/sites-ffrl/tree/v2014.4/site-rheinufer

aeickho, 2015-02-08:
I can  confirm this behavior...  it is reproduce

adlerweb, 2015-02-08:
Similar behavior on WDR4300

NeoRaider, 2015-02-08:
It is normal behavour of OpenWrt not to regard the position during the installation, only changing the switch position will have an effect. So the only issue is that the switch has its on and off positions swapped... I'll fix that next week.

mmalte, 2015-02-09:
Wouldn't it be a good idea to disable the switch at all to prevent operating errors?

NeoRaider, 2015-03-13:
Can you send me a photo of the switch? In the revision I have here the switch positions aren't explicitly labeled as on or off, I'd like to know how your version looks.

aeickho, 2015-03-14:
https://cccgoe.de/wiki/Datei:Alex_lotta_20150314.IMG_0323.JPG
Format: [image]

NeoRaider, 2015-03-15:
Fixed in git.

closed 2015-03-15 by NeoRaider