freifunk-gluon/gluon #294: Wrong model name on WNDR3700

aeickho, 2015-02-13:
On NETGEAR WNDR3700v2

The model name is  **NETGEAR NDR3700v2** and not **NETGEAR WNDR3700v2**
(cat /tmp/sysinfo/model)

this has an affect for autoupdate

tmp/sysinfo/model on (v1) is ok NETGEAR WNDR3700

348a819711fe6594bce5bc3703900871f4e2b6d3

NeoRaider, 2015-02-13:
What is the output of `cat /proc/cpuinfo` on this device?

aeickho, 2015-02-13:
root@alex_test-10:~# cat /proc/cpuinfo
system type             : Atheros AR7161 rev 2
machine                 : NETGEAR WNDR3700/WNDR3800/WNDRMAC
processor               : 0
cpu model               : MIPS 24Kc V7.4
BogoMIPS                : 452.19
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 16
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0ff8, 0x0ff8, 0x0ff8]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
shadow register sets    : 1
kscratch registers      : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

aeickho, 2015-02-13:
root@alex_test-10:~# cat /tmp/sysinfo/model 
NETGEAR NDR3700v2

NeoRaider, 2015-02-13:
Okay, could you provide a dump of your ART partition? You can find out which partition the ART is in `/proc/mtd` and then just dump it via e.g. `cat /dev/mtd3 > /tmp/art.dump`.

aeickho, 2015-02-13:
root@alex_test-10:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00050000 00010000 "u-boot"
mtd1: 00020000 00010000 "u-boot-env"
mtd2: 00f80000 00010000 "firmware"
mtd3: 00112440 00010000 "kernel"
mtd4: 00e6dbc0 00010000 "rootfs"
mtd5: 00c00000 00010000 "rootfs_data"
mtd6: 00010000 00010000 "art"
root@alex_test-10:~# cat /dev/mtd6 > /tmp/art.dump

    https://www2.artbanause.de/art.dump

NeoRaider, 2015-02-13:
I've opened a bug report upstream: https://dev.openwrt.org/ticket/18992

aeickho, 2015-03-14:
http://www2.artbanause.de/art.1.dump  one more v2 art dump

NeoRaider, 2015-03-15:
I've pushed an experimental fix to the Gluon master. Please test on as many versions of the WNDR3700v1, v2, WNDR3800, WNDRMACv1 and v2 as possible.

aeickho, 2015-03-15:
on WNDR3700v2,
cat /tmp/sysinfo/model
NETGEAR WNDR3700v2 (ok)

on WNDR3700v1
cat /tmp/sysinfo/model
NETGEAR WNDR3700 (ok)

NeoRaider, 2015-03-23:
Patch submitted upstream.

closed 2015-03-23 by NeoRaider

mmalte, 2016-02-19:
As a result of this we have some devices with:

"Model: NETGEAR"

Is is possible to identify the real model via status Page? 
http://[2a03:2260:114:4:c63d:c7ff:fe86:e7dc]/cgi-bin/status
http://[2a03:2260:114:3:a221:b7ff:feb4:9853]/cgi-bin/status

NeoRaider, 2016-02-19:
@mmalte, I think the devices wrongly reporting "Model: NETGEAR " are WNDR3700v2. The bug is fixed in both 2016.1 and the HEAD of 2015.1.x, so after updating the correct model will be shown again. The exact model doesn't matter for the upgrade, the sysupgrade images of WNDR3700v2, WNDR3800 and WNDRMAC are compatible.

mmalte, 2016-02-19:
Perfect, I'll just copy the netgear-wndr3700v2 Manifest entry an rename it to NETGEAR for our next release to catch this devices.

Thank you!