freifunk-gluon/gluon #298: Solution for the user to accept TOC and provide name

probonopd, 2015-02-22:
I know captive portals have been discussed before and deemed inacceptable, but if legislation in Germany will require (private) WLAN operators to have users accept TOC and provide their name, then we need a solution for this.

Background reading (German):
> Der Entwurf sieht also vor, dass ein offenes W-Lan nur der nutzen kann, der sich in irgendeiner Form anmeldet. Diese angemeldeten Nutzer sollen zudem einwilligen, "im Rahmen der Nutzung keine Rechtsverletzungen zu begehen" (...) zu den "zumutbaren Maßnahmen" zur Sicherung des Anschlusses auch gehört, dass sie "den Namen des Nutzers kennen"
>
> -- <cite>http://www.spiegel.de/netzwelt/netzpolitik/w-lan-bundesregierung-will-offenen-zugang-foerdern-a-1019668.html</cite>

tcatm, 2015-02-22:
I think 802.11u could provide the required functionality. Yet, this might only apply to supernodes providing gateway access to the internet. Internal freifunk access shouldn't be affected by this.

tcatm, 2015-02-22:
It looks like hostapd already supports 11u. Adding this to it's config file (I'm not sure whether OpenWRT allows this yet) should work:

    access_network_type=3
    internet=1
    network_auth_type=00http://un.freifunk.net/tos.asp

From what I've read about 11u at least Apple's device support this.

rotanid, 2016-08-22:
closing it as the current regulation doesn't need something like this.
can be reopened if this changes.

closed 2016-08-22 by rotanid