freifunk-gluon/gluon #299: Internationalization support in Gluon

Tobi042, 2015-02-22:
While preparing to deploy Freifunk in Luxembourg, the need for multilanguage support in Gluon arose. As not all Luxembourgers are fluent in German, the firmware should be multilingual, too. This might also be beneficial for asylum-seeker's hostel and people with backgrounds in different countries. The Luxembourg hackerspace Syn2Cat has offered to provide French and Luxembourgish translations and I offer to do it for English.
The way I imagine this feature to work, is either by having the firmware recognizing the browser's preferred language or by offering such an option in a prominent place in the config interface, such as the first or second element in the wizard or in the top bar.

lrnzo, 2015-03-03:
I would like to contribute here, but do not know how. Maybe i could ask some people from africa and the arabian world to do translations into there languages

spookey, 2015-03-03:
In my opinion the i18n does not stop with the config-mode. What about the status-page, the shell utilities and all those community specific plugins and scripts?

If all languages are added at once I seriously doubt that Gluon will still fit into the limited memory of the nodes.

A better approach would be a framework (or something) wich integrates the proper translation strings into the source code at compile time. It would be nice if the images of e.g. a french community are more or less the same size like those of a german one..

(Now, it pays off, that the gluon documentation was in english - well done!)

NeoRaider, 2015-03-03:
Yes, adding all language files into the images would probably not be a good idea in the long term (depending on the number of languages we actually support). But as we don't have overly much text and text is well compressable, it might be just fine.

Probably the middle way of letting the communities decide what set of languages they want to include would be the best. Most German communities will at least want to ship German and English then, I presume.

NeoRaider, 2015-03-18:
I've started adding i18n support to Gluon. I'm going to make the whole config mode translatable in the next few days and provide English texts for those pages.

I'm currently waiting for a build system patch to be merged upstream which affects the way translations are added to packages. After that, I'll put together a how-to for adding new translations and making more pages translatable.

A list of all languages supported by LuCi can be found in the file `include/package.mk`, all languages not in that list will need to be added to LuCi first before support can be added to Gluon.

NeoRaider, 2015-04-25:
I've added documentation for the i18n support: http://gluon.readthedocs.org/en/latest/dev/i18n.html

Any volunteers for making the remaining expert mode modules translatable? :D

NeoRaider, 2015-05-04:
All existing expert mode modules are translated now.

closed 2015-05-04 by NeoRaider