freifunk-gluon/gluon #303: Paralleles Freifunk Netz mit anderer SSID

RiBro, 2015-02-28:
Manche Cafes, Hotels würden gerne durch das frei angebotene WLAN werben. Da Landing Pages i.a nicht gewünscht sind würde es noch die SSID geben. Da es aber durchaus Sinn macht, dass die in einer Stadt einheitlich ist, würde ich eine Config Option vorschlagen, welche ein zweites Freifunk WLAN mit einer frei einstellbaren SSID aufspannt. 
Eigentlich genauso wie die Möglichkeit für ein Privates Netz nur das dies über das VPN geroutet wird.
Über SSH ist das leicht zu machen, ich meine aber eine Möglichkeit dies in der Config zu machen wäre nicht schlecht. 
```
>cat /etc/config/wireless 
onfig wifi-iface 'client_radio1'
	option network 'client'
	option device 'radio0'
	option mode 'ap'
	option ssid 'EigeneSSID'
	option disabled '0'

```

tcatm, 2015-02-28:
Gibt es schon Communities, die sowas machen und dafür Code haben? Dann könnte man das vielleicht integrieren. Müsste auf jedenfall jemand warten. Wenn sich niemand findet, sehe ich da schlechte Chancen, dass sowas in Gluon direkt eingebaut wird. ESSIDs sind meiner Meinung nach keine geeignete Werbefläche, wenn man bedenkt wie sehr eine ESSID bei 30 anderen Netzen in der Nähe unter geht und wie stark diese auf kleinen Handydisplays beschnitten werden.

corny, 2015-02-28:
Falls du so etwas bauen möchtest, ist mein "private-wifi"-Paket sicher eine gute Orientierungshilfe:
https://github.com/freifunk-gluon/packages/tree/master/gluon/gluon-luci-private-wifi

Ich glaube allerdings nicht, dass sich viele Freifunker dafür begeistern können. Ich sehe es lieber, wenn Cafés/Hotels Werbung für Freifunk machen und hiedurch die Bekanntheit von Freifunk gesteigert wird.

mmalte, 2015-03-01:
Dabei sollte auch bedacht werden dass wir durch das zusätzliche Mesh bereits 2 SSIDs ausstrahlen, eine dritte verbraucht noch mehr Bandbreite, siehe dazu auch "Multi-SSID Deployment Considerations":

"As the number of wireless networks operating on a specific channel increase so does the amount of beacon frames and probe responses. Take a scenario where there are two physical APs on the same channel each with a single SSID. Both APs will transmit one beacon frame every 100 ms and when any client sends a probe request on that channel, each AP will send a probe response. This would not cause much overhead. However take the same two physical APs each with 4 SSIDs. Now 8 VAPs are independently sending beacon frames every 100ms and any time a client sends a probe request, 8 probe responses are transmitted."

Quelle:
https://kb.meraki.com/knowledge_base/multi-ssid-deployment-considerations

A-Kasper, 2015-03-02:
Durch nicht konstante SSIDs geht die Roaming Möglichkeit verloren. Ich
würde das nicht unterstützen.

tcatm, 2015-03-02:
Ich mach das mal zu. Falls ihr sowas braucht, könnt ihr das ja in einem [Community Repo][1] entwickeln und erproben.

[1]: http://gluon.readthedocs.org/en/v2014.4/user/site.html#packages

closed 2015-03-02 by tcatm