freifunk-gluon/gluon #304: Add an option to have more ethernet ports in the WAN network instead of the Freifunk network

eriktews, 2015-03-01:
Hi

Assuming that you would like to provide Internet to a location where one AP cannot cover the whole location. The setup should look like a private ethernet network and private WiFi (WPA) for the whole location and Freifunk should be supported as well. Let's call the private network just WAN.

It would be great when the local ethernet ports of the device or at least some of them could be bridged with the WAN network, since this would allow me to do it with only a single cable to every AP. It can then provide wired internet for wired clients who would like to connect to the WAN network as well as WiFi for Freifunk and the internal WiFi. Currently, I need either a second cable to each location for wired clients or I need to add an additional hardware switch to the location so that I have more ports for the WAN network.

Currently, I can do that manually by logging in and changing the OpenWRT switch configuration. It would be great when there would be an option in the configuration page to do this.

Since the logical switch port numbering used in OpenWRT is not always the same as the labels printed on the device, I suggest to make an easy switch for changing all the switch ports from Freifunk to WAN, and maybe have a more hidden "expert mode" setting where you can change the status of each switch port individually.

A-Kasper, 2015-03-02:
I highly agree with this request. In my case I've got some CPE210 and
CPE510 outdoor devices. They have got a range of about 120°. I would
like to put 2 of each side-by-side  to cover a big place from a roof.
I've got one CAT5 cable going there. I would really like to chain them
with the two LAN ports they've got, so I don't need to worry about how
to make a switch weatherproof. The requested feature would be very
useful for such setups.

tcatm, 2015-03-02:
This is basically mesh-on-lan, isn't? Support for mesh-on-lan is in https://github.com/freifunk-gluon/packages/pull/63 and will be possible in the next release.

eriktews, 2015-03-02:
I'm not so sure, but after I read through other bug reports, I assume that the correct name feature will be called **uplink on lan**. Is it the same as mesh on lan?

But in addition to "uplink on lan", it would also be wise to enable meshing on the wan interface so that the devices can exchange their mesh traffic on the cable and don't need to use the air bandwidth.

NeoRaider, 2015-03-02:
Changing all LAN ports to WAN will definitely be supported by #284, which I plan to work on very soon (I'm currently busy with exams...)

Configuring distinct ports of a VLAN-capable switch to different networks is also planned in the long term, but will require more work as different hardware requires different handling here.

eriktews, 2015-03-02:
OK, after having thought about this a bit more, I think a very useful option would be just to swap LAN with WAN ports. As a result, you have 4 LAN ports (on most devices) for the same ethernet as the uplink network, and a single port is still in the client network, so that you can still have wired access to the freifunk network. Since the WAN and LAN ports are easy to distinguish visually, there should be less problems than when changing every port of the switch manually.

tcatm, 2015-03-03:
I'm closing this as  the functionality will be covered by #284

closed 2015-03-03 by tcatm