freifunk-gluon/gluon #306: Images for virtual VPN offloaders (KVM/VBox) lack network drivers

MPW1412, 2015-03-06:
Hello,

I tested the images for KVM/Qemu and VirtualBox. Both don't have the correct drivers to use the virtual network adapters.

Any help to include them will be appreciated. I never compiled a kernel myself nor do I know where to active the modules in the gluon git folder.

Any ideas?

Greetings
MPW

NeoRaider, 2015-03-06:
The modules just have to be included by adding the needed packages to the GluonProfile lines in `targets/x86-generic/profiles.mk` (like it is done with the kmod-pcnet32 module there; multiple packages must be separated by spaces).

According to https://blogs.oracle.com/fatbloke/entry/networking_in_virtualbox1 , the default network adapter in VirtualBox is a AMD PCNet FAST III (Am79C973), which should work with the pcnet32 driver. Has this changed and VirtualBox is now defaulting to virtio-net as well? Does VirtualBox make use of any other virtio drivers (virtio-balloon or virtio-random)?

Well, I guess adding a few more drivers to the generic image won't hurt anyways, OpenWrt includes the following by default: kmod-3c59x kmod-e100 kmod-e1000 kmod-natsemi kmod-ne2k-pci kmod-pcnet32 kmod-8139too kmod-r8169 kmod-sis900 kmod-tg3 kmod-via-rhine kmod-via-velocity. I'll add these and the virtio modules to the generic build.

MPW1412, 2015-03-06:
I don't see any interfaces except lo. How do I bring the interfaces up? Did you test it in bridged mode?

NeoRaider, 2015-03-07:
That doesn't answer my question at all. I was asking which interface type is selected in the VirtualBox settings by default. According to https://blogs.oracle.com/fatbloke/entry/networking_in_virtualbox1 there are three options (e1000, pcnet32 and virtio), but that page is three years old...

I've never tested any of the x86 images except the VMware one, which is working fine.

MPW1412, 2015-03-07:
Ah sorry. I tried different network adapters and PCnet-FAST II+III work. 

I have a devices named eth0 and br-setup now. But they don't ask for addresses from my FritzBox.

br-setup has the ip 192.168.1.1. And eth0 has no ip-address. But my FritzBox takes at least notice of the new device and lists it as follows: 

PC-08-00-27-4C-87-78	-	08:00:27:4C:87:78	LAN 1 mit 1 Gbit/s
(the mac matches the eth0 device)

Maybe the system is stuck in configuration mode?

NeoRaider, 2015-03-07:
Network drivers added, other issues handled in IRC.

closed 2015-03-07 by NeoRaider