freifunk-gluon/gluon #322: Weird bridge behaviour

NeoRaider, 2015-03-31:
We've had multiple reports of weird bridge behaviour with Gluon v2014.4, for example:
* `br-wan` stops responding to neighbour solicitations occasionally
* `br-wan` doesn't forward DHCP correctly when the LAN ports are moved from `br-lan` to `br-wan` occasionally

The neighbour discovery issue might be caused by multicast issues, but that doesn't really explain the DHCP problem. Another possible cause might be be netfilter/conntrack bugs.

T-X, 2015-04-10:
I can confirm this issue and it's quite easy to reproduce:

Take a recent stable gluon version. Look up the ipv6 link-local address on br-wan (ip -6 a s dev br-wan). Try to ping6 that address from a host connected to the WAN port - it will fail.

On a 4.0.0-rc3 kernel I couldn't reproduce that issue. The following things are workarounds on gluon after which IPv6 seems to work again (any one works):

* echo 0 > /sys/class/net/br-wan/bridge/multicast_snooping
* ifconfig br-wan promisc
* /etc/init.d/firewall stop and wait a few seconds (probably up to 125 seconds until the next MLD report)

Especially as the last one is a valid workaround I suspect your last assumption might be the right one. So that netfilter is preventing multicast reports to reach the bridge kernel module.

NeoRaider, 2015-04-10:
Okay, the first one seems like the most workable solution.

@T-X, did you also have a look at the DHCP issue, which shouldn't involve multicast at all, but only broadcast?

T-X, 2015-04-18:
Just noticed, that there's never been a gluon release with this commit yet:

"kernel: bridge: multicast: backport a few more fixes for 3.10"
(http://git.openwrt.org/?p=14.07/openwrt.git;a=commit;h=60abf1305e4c8cc8c5673dfd98b214e9ceab8ca5)

I'm going to ask THiSCO whether the issue still exists for him with a recent gluon-master build.

thisco-de, 2015-04-20:
@T-X I first noticed the problem with a Gluon build from master built some time around the beginning or mid of March and then reproduced it with 2014.4. As the patch you mention dates back to January, it should thus be included in the build (at least the patch files exist in my build-dir).

Nevertheless, for those who want to (temporarily) disable multicast snooping on br-wan, at https://gist.github.com/thisco-de/eb49a9fb216c9adb87dd there is an update-script (should be placed in `/etc/uci-defaults/`), which at least has proven to workaround the problem reliably on my node. Probably @NeoRaider even wants to include this in Gluon until the problem has finally been solved.

T-X, 2015-04-20:
Upgraded the node I could confirm the issue with to a gluon experimental branch which has that commit. And yes, can confirm that the issue is still present there.

Going to ask whether that rings a bell for someone on the bridge mailinglist. Otherwise need to refresh my memory about OpenWRT/gluon/quilt I guess, to get some debug output in :).

yanosz, 2015-04-20:
@T-X and I discussed similar issues for the KBU-firmware ("Gluon flavored OpenWRT").

Like gluon, there's br-freifunk, having bat0 + AP wifi's in it. We noticed that:
* For 14.07 (Release) + batman-adv-2014.4.4 (OpenWRT-Routing, master, as of today)
 *  ND for br-freifunk were not leaving bat0 - and thus didn't succeed.
 * After some minutes they did (@T-X consideres ND to go Unicast, then).
 * Stopping firewall, firewall-ebtables, unloading modules had no effect
 * Disabling multicast_snoiping did not help
 * When stored in the neighbor cache (eg. by ping's vice versa) everything is fine.
* For 14.07 (branch, as of today)
 * Problem existed
 * Disabling ICMP snooping fixed the issue instantly.

As of today we're considering to ship <code>uci set network.freifunk.igmp_snooping=0</code>.
@T-X has some pcap-Traces of this issue

NeoRaider, 2015-04-27:
We've now disabled `igmp_snooping` on the WAN bridge.

T-X, 2015-04-29:
Oki doki, makes sense for now as the real issue seems to be a little harder to track down. So, like you asked on IRC, moving the milestone to 2015.2 is fine by me.

While it's easy to reproduce with the gluon firmware, I couldn't reproduce it in VMs with 3.12 or 3.10 or 3.10.49 + OpenWRT-bridge-patches yet.

I'm going to try a few more things... (like applying the netfilter patches in OpenWRT to the vanilla 3.10.49, too, blacklisting netfilter modules in Gluon, building Gluon without any netfilter modules and finally some printk()'s for the bridge in Gluon).

T-X, 2015-04-29:
A few more interesting findings for the neighbor reachability issue:

* Moving /lib/modules/3.10.49/ebt* out of there does not help, but
* Moving /lib/modules/3.10.49/ip6tables_filter.ko out of there helps
* Setting "option conntrack" to 0 in the wan zone helps

T-X, 2015-05-01:
Okay, "Setting "option conntrack" to 0 in the wan zone helps" was actually a false positive. It just took a little longer than usual at that try.

@NeoRaider, the good news is, it seems like my other assumption I told you about a few hours ago seems to be it :). It's simply a broken firewall setting, not an issue in the kernel, yaiy!

Adding the following sections works for me, according to the iptables/ip6tables counters they seem to match:

<pre><code>
config rule              
        option name 'Allow-IGMP'
        option src 'wan'     
        option proto 'igmp'   
        option family 'ipv4'   
        option target 'ACCEPT'

config rule                      
        option name 'Allow-HOPBYHOP-Input'
        option src 'wan'          
        option proto '0'      
        option src_ip 'fe80::/10'
        option family 'ipv6'  
        option target 'ACCEPT'
</code></pre>

(unfortunately making more specific MLD rules instead of the general hop-by-hop-option rule seems impossible as ip6tables/netfilter cannot match protocols behind options)

NeoRaider, 2015-05-01:
Hmm, the hop-by-hop match is a bit problematic. Wouldn't that basically allow any WAN input by just prepending a hop-by-hop header before the actual content?

Apart from that, does this also affect upstream? If it does, can you send a bug report / patch there?

Unfortunately, this still doesn't explain the other issue I mentioned in this issue, the lost DHCP packets, at all.

T-X, 2015-05-01:
Yes, that's also what I don't quite like about it, that you could easily insert a hop-by-hop header. But at least you shouldn't be able to exploit that over the internet, that's why I added the check for the link-local address.

Yes, will check later, whether it affects upstream.

For DHCPv6, can't tell, didn't check yet and didn't see any captures of that etc. Are you sure it's persistent? It might be, that after such a move you might need to wait for 125 or 255 seconds for the next MLD query or something like that.

NeoRaider, 2015-05-01:
This is about DHCPv4.

T-X, 2015-05-05:
Actually, the MLD specific rules work if you specify the icmpv6 code of zero explicitly. There seems to be a bug in OpenWRT that if you leave it unspecified that it gets set to 255.

The solution that got upstream is this one:

https://dev.openwrt.org/changeset/45613

For DHCPv4 I'd suggest that the person experiencing this issue should open a new ticket with a more specific title, a description on how to reproduce it and some tcpdump captures.

NeoRaider, 2015-05-28:
See also: See also: https://bugzilla.kernel.org/show_bug.cgi?id=99081

NeoRaider, 2015-10-11:
As the original bug (bridge multicast issues) has been workarounded by disabling the multicast snooping, I'm considering this issue fixed.

Further work is being prepared in #402, but will not necessarily make it into Gluon 2015.2.

closed 2015-10-11 by NeoRaider