freifunk-gluon/gluon #328: [enhancement] (Nano m2/m5) PoE passthru via site.conf or config-mode?

RubenKelevra, 2015-04-03:
Currently the command to enable PoE-Passthru is very cryptic, maybe we should add an option to enable this function via config-mode.

chris007de, 2015-05-18:
Regarding the questions whether to use site.conf or config-mode, I would strongly recommend to use the config mode, just because of convenience for the users.

NeoRaider, 2015-05-18:
We have many site.conf options which aren't supposed to ever be set in "normal" firmwares (like mesh_on_wan, authorized_keys, and a few others). It is still nice to have them to allow completely preconfiguring images for special installations. Of course, such settings should also be added to the config mode.

tcatm, 2015-07-17:
This OpenWrt commit is relevant to this issue: http://git.openwrt.org/?p=openwrt.git;a=commit;h=24d54f8b0c6b18e5ec1f8c202afdbae14d347737

To my knowledge we'd want to at least support Nanostations, CPE210 and CPE510. Are there other devices supporting PoE passthrough?

SimJoSt, 2016-04-13:
I would consider adding this option only after #245 has been resolved. At the moment everybody uses PoE-passthrough at their own risk. Making it easier to use this option would create the need for a warning message next to the option.

closed 2016-07-29 by NeoRaider

4ndr3, 2016-10-12:
Klasse Funktion, habe ich getestet, läuft prima! Drei bitten habe ich noch:

1. Bitte die entsprechenden uci-Befehle zum de-/aktivieren in der Doku hinzufügen: https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#poe-passthrough
2. Bitte einen warneden Text dran schreiben, dass an den LAN-Port angeschlossende Geräte 24V passives PoE abkönnen müssen oder sonst kaputt gehen
3. Das ganze steht im config mode derzeit unter der Überschrift WAN Connection. Über die Funktion müsste, um Verwirrung auszuschliessen "LAN" stehen. Schliesslich wird es das PoE durch den LAN Port durchgereicht (in der original firmware LAN1 oder secondary LAN port genannt)

Besten Dank!

commit f3cb6d7b (2016-07-29): gluon-luci-portconfig: add setting for PoE passthrough

Fixes #328