freifunk-gluon/gluon #331: Add node-groups to show which nodes belong together

oakey-dev, 2015-04-07:
until now one usually names the nodes which belong together with [prefix]-[nr] or so,
but wouldn't it be better if one has a "Group" field within the config-mode.
There one can enter a name, which then is announced in alfred so than one can group nodes by that name?

NeoRaider, 2015-07-12:
I think such a feature might confuse normal users.

jplitza, 2015-09-19:
I don't see any reason why that would be better than having the prefix in the name. After all, such a "group" isn't guaranteed to have *any* meaning. It could mean the nodes all owned by the same person (we have the owner information for that), they are all installed in the same area (we have location information for that) or something completely different.
Everything except for the name that the nodes announce should at least have some well-defined meaning, and this field doesn't have that as far as I can see.

closed 2016-08-22 by rotanid