freifunk-gluon/gluon #332: Android doesn't connect if the node has no internet access

rubo77, 2015-04-07:
The idea of freifunk is that it works without internet access also as a local mesh cloud.

Why does my android phone not connect to the freifunk net if it is split from the internet?

Couldn't we somehow meke this work?

adlerweb, 2015-04-07:
Hard to tell without details of your setup. Most likely you have a central DHCP-server so your android doesn't get an IP and considers the network as dead. Most other sysems fall back to APIPA/ZeroConf in this case, however this is afaik not implemented on Android. Workarround would be to enter a IP manually on your android phone or set up a local DHCP server on the routers, however the latter would require some effort so prevent duplicate IPs or interfering with central internet gateways.

do9xe, 2015-04-08:
The actuall probleme here is on the android-side. In the older versions Android is not able to obtain an IPv6 Address. It works in the newer Versions, but it is not able to connect to a v6-only network.
We have talked about decentralized DHCP before, but it is hard to realize. We could talk about this at ffnord-con.
Or just open a bug-report at android ;)

NeoRaider, 2015-07-12:
Closing, as it is an Android issue, and decentralized DHCP is currently in development.

closed 2015-07-12 by NeoRaider