freifunk-gluon/gluon #333: empty error message when choosing wrong hostname

rubo77, 2015-04-09:
If I choose a hostname with special characters öäü in it and press the commit button, the page reloads and shows an empty error message (with just a list-dot and no text).

NeoRaider, 2015-04-09:
Ah, I might have broken that when I cleaned up the templates for the i18n support. I'll look at it.

rubo77, 2015-04-09:
[image]

NeoRaider, 2015-04-18:
Ah, it's the other way around: It was broken before, I fixed it (unknowingly) when I added the i18n support. So this bug had already been fixed when you reported it.

Please always mention the Gluon version you are using in your bug reports.

closed 2015-04-18 by NeoRaider