freifunk-gluon/gluon #334: neighbour log over time

rubo77, 2015-04-11:
Especially moving mobile nodes but also for nodes with neighbours with differing connection strength it would be of most interest to which nodes they connect over time.

My first solution idea:  
We would need a log of all neighbours seen over time, preferingy only local in RAM, that is sent through the mesh once the node has connection to a definable target address that will collect the connection information.

ohrensessel, 2015-05-07:
mh you could log the information available in the nodes.json file generated by the ffmap-backend to see the mesh partners of a node over time.

rubo77, 2016-08-07:
But this would be interesting on the status page of a node too, so a history of seen nodes would fit in there

mweinelt, 2018-11-14:
Is what the current status page provides sufficient? 

Fwiw, I don't think it's feasible to keep any history on the router.

rubo77, 2018-11-14:
the history should not be written on the router, but nodes that are there and gone again, should be visible until reboot. At the moment nodes are only visible for a certain time on the right (how long?)

mweinelt, 2018-11-14:
In your opinion, where should this information be kept if not one the router?

rubo77, 2018-11-15:
The history could be kept in a temporary file

rubo77, 2018-11-15:
Sie, in memory, but that would be only a few bytes. And we don't want to cause write cycles

mweinelt, 2018-11-15:
That would require a cronjob/daemon to collect the information regularly, and I'm telling you I don't like the sound of it, because I think the gain is very minimal.

rotanid, 2018-11-15:
ack and therefore rejected

closed 2018-11-15 by rotanid