freifunk-gluon/gluon #336: HT40 is not possible for 5 GHz Mesh.

RubenKelevra, 2015-04-11:
We tried the following site-conf with 2014.4.x:

<pre>
wifi5 = {
		ssid = 'lev.freifunk.net',
		channel = 44,
		htmode = 'HT40-',
		mesh_ssid = 'mesh-v15',
		mesh_bssid = '02:ca:fe:15:21:44',
		mesh_mcast_rate = 12000,
},
</pre>

Which fails with

<pre>
root@lev-X:~# logread | grep -E "radio1|mesh1" | grep "fail"
Sat Apr 11 11:02:25 2015 daemon.notice netifd: radio1 (1190): command failed: Device or resource busy (-16)
</pre>

In iw dev the device appears, but not with a configured channel,  it seems that the HT40 channel-definition was not applied correctly:

<pre>
root@lev-X:~# iw dev
phy#1
    Interface client1
        ifindex 20
        wdev 0x100000005
        addr 12:00:ef:40:a8:20
        ssid lev.freifunk.net
        type AP
        channel 44 (5220 MHz), width: 40 MHz, center1: 5230 MHz
    Interface mesh1
        ifindex 18
        wdev 0x100000004
        addr 12:01:ef:40:a8:20
        type IBSS
phy#0
    Interface client0
        ifindex 19
        wdev 0x5
        addr 12:00:ee:40:a8:20
        ssid lev.freifunk.net
        type AP
        channel 5 (2432 MHz), width: 40 MHz, center1: 2422 MHz
    Interface mesh0
        ifindex 17
        wdev 0x4
        addr 12:01:ee:40:a8:20
        ssid mesh-v15
        type IBSS
        channel 5 (2432 MHz), width: 40 MHz, center1: 2422 MHz
</pre>

This problem was observed here too:
https://github.com/freifunk-berlin/firmware/issues/160

NeoRaider, 2015-04-11:
Well, obviously `HT40-` is wrong for channel 44 as 44 is already the lowest possible channel. You need to use `HT40+`.

closed 2015-04-11 by NeoRaider

RubenKelevra, 2015-04-11:
channel 36, 40 are not allowed to be used by mesh-networks?

RubenKelevra, 2015-07-29:
@NeoRaider This Problem has been fixed on upstream.

https://dev.openwrt.org/ticket/18268

NeoRaider, 2015-08-04:
1. Obviously, I was wrong about channel 44 being the lowest possible...
2. The ticket you mentioned seems to recommend using noscan? Also, I don't see anything fixed there...
3. Your original issue is identical to #386 , which is now fixed in Gluon.

RubenKelevra, 2015-08-13:
In our communitys we changed the channel for 5 GHz now anyway, just wanted to make sure that the issue is solved. Thank you!