freifunk-gluon/gluon #338: Feature-Req

RubenKelevra, 2015-04-16:
I have often read, that some freifunk-communtiys explain that, you can change the QoS-Settings in e.g. your fritzbox, for moving fastd-traffic in background.

Since this requires a deep understanding of ports and firewall-settings, it might be a good solution to tag fastd-traffic with a tos-bit, like we already do on our gateways.

To move the traffic to background-tag, it should be enough to set the following rule:

-A OUTPUT -p udp -m mark --mark 0x1 -j TOS --set-tos 2

ohrensessel, 2015-05-07:
is it known if any normal routers for private use evaluate TOS? does the fritzbox do that?

RubenKelevra, 2015-05-08:
Well, I know that any Linux does ... the QDisc pfifo_fast uses ToS for building different queues:

See also:
http://linux.die.net/man/8/tc-pfifo_fast

My asus-router uses pfifo_fast as default qdisc.

Since nearly all boxes does VoIP too, they should support ToS as well, especially Fritz!Boxes.

RubenKelevra, 2015-05-11:
Support is confused as usual, but fritz!boxes seem to support this as well.

https://twitter.com/rubenkelevra/status/597761661638938624

T-X, 2015-05-11:
Hm, background tag. I actually wanted to ask for a fastd feature to inherit the QoS tag from the inner packet. To make things like VoIP at least a little useable in a Freifunk network, for instance.

ohrensessel, 2015-05-11:
mh their answer seems quite clear to me. They do not evaluate the ToS bits
but support QoS by mapping certain IP/Port combinations to three different
classes via the webinterface (thats what I'm using in my fritzbox).

RubenKelevra, 2015-05-11:
@T-X so you want to parse the dscp-field from the ip packet behind the ethernet+batman-adv+ethernet header?

Since then it's still interesting to define a default-class for the fastd packages, especially if the freifunk is used heavly on small dsl-upstreams. Background just moves the freifunk-traffic in the priority behind the traffic of the internet-contributors - which we should, because it's fair.

@ohrensessel Sure, the answer is fully unclear. But fritzboxes should support the dscp-fields for routed / forwarded traffic.

ohrensessel, 2015-05-15:
@RubenKelevra I don't see why they **should** support dscp fields. the answer from the support does not state anything like that. and their options in the webinterface use a totally different technique: traffic classes per protocol, port and ip combination.

I think that no or only a small minority of home dsl routers supports evaluating dscp fields. If you have more detailled information which routers have support for that, I would be very happy.

RubenKelevra, 2015-05-16:
Fritz!Boxes have QoS support:
http://www.wehavemorefun.de/fritzbox/Qos

As I already said, Linux have a default-support for QoS-Classes, so all Home-Routers which runs a Linux-Kernel supports QoS.

Since nearly everybody are talking about "VoIP-Support" or "Quality of Service" in their Ads, I think nearly every home-router support different queues for different dscp-flags.

Since this flag does not break anything, and only use the feature, when they are available, and does not cost any noticeable cpu performance, but might increase the network-response for the homeuser, which spends their internet-connection, we should flag our packets like that.

The only argument against it is a bit more jitter on high loads on the private networks. But I think this might also help batman to detect overloaded links over vpn, so the traffic might flow to other vpn connections.

NeoRaider, 2015-07-12:
@T-X, do batman-adv packets inherit the tag from the inner packet?

NeoRaider, 2015-10-18:
I just noticed that my last question didn't make sense, as the ToS/DSCP is an IP header field, not an Ethernet flag.

The Wikipedia article on DiffServ looks like the definition of the field has changed much too often to be of any use... the Linux headers (and implementation?) still seem to be based on the previous "ToS" definition that is obsolete since 1998 or something...

Is this field actually used by any real-world software in the internet? I.e., do current VoIP applications set the flag on their packets? It sounds more like something used in corporate networks and other controlled environments. Also, Wikipedia states that the actual encoded values of the DSCP field may be implementation-defined, and the RFCs only make recommendations how to use the field...

Btw.: http://www.wehavemorefun.de/fritzbox/Qos rather sounds like the opposite of "Fritz!Boxes have QoS support": there are some commands on the FritzBox that have something to do with QoS, but important parts are missing, so it is non-functional.

RubenKelevra, 2015-10-19:
Well I think the main issue is that no standard except ToS has the ability to flag traffic with a background-tag. So you first have to redefine one field as background-traffic.

Since ToS blocks the use of ECN, I think it is a very bad idea to use ToS at all.

bittorf, 2015-10-19:
hmmm...a alot of dangerous superficial knowledge 8-)
"The DSCP and ToS bytes are actually the same byte. ToS is from a pre-1998 RFC, and DSCP is after that. The important bits overlap pretty awkwardly:"
```
DSCP: PPPNNNEE
P = Forwarding level/Precedence | N = DSCP level | E = Error Congestion Detection
ToS:  PPPNNNN-
P = Precedence | ToS value | Ignored, but counts (so 0x10 is the leftmost N)
```
see:
https://github.com/bittorf/kalua/blob/master/openwrt-addons/etc/kalua/netfilter#L2235

Not every linux-box does evaluate/honour the ToS bits (or DSCP) from outgoing packets - It's the packet-scheduler of your interface which does it, e.g. pfifo-fast does it. fq_codel does not. If you want it and only have e.g. fq_codel on your WAN-interface, you must stick an additional queue on your interface.

RubenKelevra, 2015-10-19:
Well, just wondering why do you think this is not known. We just discuss about which standard is commonly implemented in latest linux based hardware. I think pfifo-fast is still very common, at least my Asus-Router use it.

The issue is, that we can't flag something as background with DSCP and using ToS is very odd because it is not compatible with ECN

bittorf, 2015-10-19:
so the asus is an old router of your ISP? hmmm...so you must distinguish between setting prio in mesh and for "further" routing...

NeoRaider, 2015-10-19:
I guess adding an option to set the ToS flag in fastd would actually make sense, as pfifo_fast is indeed the default qdisc and many (most?) Linux-based routers use it.

The conflict with ECN doesn't matter, as ECN is not used by fastd anyways (ECN is only really used for TCP usually). I'd even say it's the other way round: using ECN is a bad idea even for TCP, as it conflicts with ToS, and there are still too many pfifo_fast qdiscs around to make its use safe.

@bittorf, this is not about the ISP's routers. What matters is the home router of a Freifunk user, making it handle fastd traffic with lower priority than the traffic from the users private network sound like a good idea. I don't think any ISP routers do handle, or should handle, ToS.

RubenKelevra, 2015-10-19:
@Bitdorf I do not want a router from my isp. I have a modem. The Asus-Router is the latest highes model of the N-Standard-Devices from Asus.

We never talked about prio in mesh you must misinterpreted something here. Read again.

bittorf, 2015-10-19:
@RubenKelevra my name is 'Bittorf': If the Asus-Router is "latest", does it have fq_codel or pfifo_fast as scheduler? Does it run OpenWrt?

closed 2022-06-17 by RubenKelevra