freifunk-gluon/gluon #339: Feature-Req: Redirect automatically from factory firmware to setup-mode

RubenKelevra, 2015-04-16:
Currently, after flashing the factory-firmware, the page does not reload. All setup-a-freifunk-node-guides now explain the users, that they have to type 192.168.1.1 now in the address bar.

Since we have a http-server running, and knows which hardware where booting, we can send a http-redirect from the wrong ip to the right one: 192.168.1.1.

So for ubiquity-hardware we need to listen and redirect from 192.168.1.20, for TP-Link-hardware its 192.168.0.1.

This would improve the end-user experience and speed up the flashing of many nodes as well.

rubo77, 2015-04-17:
Good Idea, I thought about this already also, but the node would have to serve a http-redirect response on 192.168.0.1 and 192.168.1.20 then apart from serving the config mode sites on 192.168.1.1, is this even possible?

ohrensessel, 2015-05-15:
doesn't this idea include patching the factory firmware? because its the one that needs to send a redirect. therefore: not possible?

rubo77, 2015-05-18:
Can we reopen this?

I think, we could solve this without patching the factory firmware:

The node could serve on 192.168.0.1 a second page that **only redirects** to 192.168.1.1where the config mode dwells

NeoRaider, 2015-05-18:
@rubo77, this ticket *is* open.

RubenKelevra, 2015-06-20:
@ohrensessel no, it's just necessary to configure more than one ip-address and let the http-server listen to 0.0.0.0 and let it do redirecty on everything else than 192.168.1.1 *to* 192.168.1.1.

Adding some IPs to the network-interface shouldn't be difficult, like
 192.168.1.254 for TPLink Wireless-AP-Devices
 192.168.1.20 for Nanostations

jplitza, 2015-09-19:
Redirecting cross-address will not work for all those addresses, because the client has configured *either* 192.168.0.0/24 *or* 192.168.1.0/24.

You could of course try to simply serve the setup-mode on all desired IP addresses and redirect from every unknown URI to the start of the setup-mode. That can be accomplished easily by passing `-E /index.html` to uhttpd (making it serve the index.html, which already does the redirect, as 404 document).

If that is the way to go, all that should be left to do is
* add the desired IP addresses to package/gluon-setup-mode/files/lib/gluon/setup-mode/rc.d/S20network (and K90network)
* add the `-E` switch to package/gluon-setup-mode/files/lib/gluon/setup-mode/rc.d/S50uhttpd and 
* adjust the DHCP range in package/gluon-setup-mode/files/lib/gluon/setup-mode/rc.d/S60dnsmasq

RubenKelevra, 2015-09-19:
Well sure, your right. Serving the config-page on all IPs would be working better. My intention was, that we could just route, because the most firmwares serve it's own IP as router. But this won't work on ubnt hardware nor on TP-Link ap-hardware.

jplitza, 2015-09-20:
That's another interesting aspect: The default route. When the client uses the DHCP server that is started in setup-mode, it doesn't get a default route, which is intended so that things like #58 could possibly be using another, wireless internet connection the client has. But If we do not force the client to re-request a DHCP lease, the default route prevails.

I'm not sure if this is a problem. We currently do not use the theoretical existence of the internet connection, but it would be lost with this change.

NeoRaider, 2015-09-20:
The client will make a new DHCP request when the link gets lost during the upgrade... and if that is the case is probably hardware-dependent.

We'd definitely have to change the DHCP range and address prefix to 192.168.0.0/16, so all 192.168.X.0/24 networks are covered. That is probably fine, but I don't really like it. More addresses and bigger address ranges also increase the probability of collisions with a potential other (wireless) connection on the client...

rubo77, 2015-09-20:
I thought the only collision would be, if another network already uses exactly 192.168.1.1.

If we add this feature there are exactly and only three more possible collisions: 192.168.0.1 and 192.168.1.20 and 192.168.1.254

Can you give an example of other collision that could occur if we change it?

RubenKelevra, 2015-09-21:
@jplitza the wirelink get lost for a very short period of time, so the os of the pc seem to not reach the threshold for detecting a connection-change.

NeoRaider, 2015-09-23:
@rubo77, all overlapping prefixes do collide in some way. If a host is connected to two networks, for example,

* Network A: 192.168.1.0/24
* Network B: 192.168.0.0/16

then the host won't be able to reach any host in the network B that uses an address of the form 192.168.1.x, as these addresses are routed to network A instead.

If both networks use the same prefix (e.g. 192.168.1.0/24), then the behaviour is completely undefined and depends on the host's operating system. It will only be possible to contact the hosts of one of the networks, but not both.

The only way to *really* fix this would be to use an IPv6 ULA for the setup mode as well, as the limited private IPv4 address space makes it impossible to effectively avoid such collisions. I don't have a good plan how to do this nicely though at the moment, so we'll have to live with such collisions for a while. After all, the possiblity of connecting to both the node in config mode and a local network at the same time is rarely used anyways, so this isn't a problem normally.