freifunk-gluon/gluon #340: Txpower cannot be set lower than 15 dBm at Nanostation M2.

christian1980nrw, 2015-04-20:
Tested @gluon-moehne-0.6.2-stable20150415-ubiquiti-nanostation-m-sysupgrade.bin

uci set wireless.radio0.txpower=8
uci commit
wifi
iwinfo wlan0 txpower

The result will be 20 dBm, in Germany you are only allowed to use 8 dBm with a Nanostation M2.

uci set wireless.radio0.txpower=0
The result will be 15 dBm. Lower settings are not possible on a M2 (without Loco).

On a Nanostation M2 Loco it is working fine and you can set every value from 0-20 dBm.
Please confirm and fix this bug asap. Thanks.

tcatm, 2015-04-20:
Can you confirm this to be a duplicate of https://github.com/freifunk-gluon/gluon/issues/94?

gomaaz, 2015-04-20:
confirming issue-description

christian1980nrw, 2015-04-20:
Issue #94 seems to be similar but i dont think its a duplicate.
The Txpowerlist doesnt have enough rows/columns.
Unfortunately, the use of gluon on a M2 is illegal in germany because of this bug.
I will turn off my hardware until this is fixed.

tcatm, 2015-04-20:
I've just tried it on a Nanostation M2:
```
uci set wireless.radio0.txpower=8
uci commit
wifi
iwinfo wlan0 txpower
```

It will set the output power to 20dBm which equals 100mW and is perfectly fine, even in Germany.

christian1980nrw, 2015-04-20:
Thanks for your reply @tcatm . Please compare it with a M2 Loco. Here you can see that the antenna gain is not considered. You have to go lower to 8 dBm.

tcatm, 2015-04-20:
Could you post both `iwinfo wlan0 info` and `iwinfo wlan0 txpower` for both your devices with txpower set to 8?

christian1980nrw, 2015-04-21:
The Txpowerlist doesnt have enough rows/columns so you cant go belower 12 on a M2.
[image]

tcatm, 2015-04-21:
I see it's already at 20dBm for the Nanostation M2 so I don't see the problem. Have you checked `iwinfo wlan0` info? It should tell you the tx chain gain (12dB). `iwinfo wlan0 txpower` will take that into account.

christian1980nrw, 2015-04-21:
Youre right. The M2 Loco has no TX power offset.
Could it be integrated into the firmware?

Nanostation M2 Loco:
Hardware:  [Generic MAC80211]
TX power offset: unknown
Frequency offset: unknown

Nanostation M2:
Hardware: [Ubiquiti NanoStation M2]
TX power offset: 12 dB
Frequency offset: none

tcatm, 2015-04-21:
I think you'd need to add it to http://git.openwrt.org/?p=project/iwinfo.git;a=blob;f=hardware.txt;h=4cac0a1258e266e6d2f5de0f601274f3cd11e31a;hb=HEAD

Can you do that?

christian1980nrw, 2015-04-21:
Thanks. So its not a bug, its a missing Hardware-Description in the firmware at the moment. So i can set the Locos to 11dbm (without Antenna gain) and the M2 to 20dBm (with anetenna gain in the firmware). Closed.

closed 2015-04-21 by christian1980nrw