freifunk-gluon/gluon #342: Add chatbox/shoutbox to status page

peki75, 2015-04-21:
Add simple chatbox/shoutbox to status page to allow basic communication (even if node is isolated from mesh).

Chat will be in memory only, when node is restarted the history is gone. Also the memory for the chat history should be minimal since we're working with limited devices. This could be also some kind of guestbook when people are happy to find a Freifunk node and want to say "Thank you!".

dracoTrier, 2015-05-13:
I hope gluon doesn't turn into a bloatware... Please keep it lightweight and simple! Routers should do routing/bridging, meshing and basic firewalling (for multicast/broadcast-stuff) but not much more.

MichaelSasser, 2015-08-26:
If you have one node this is ok, but if you have many nodes this is weird. A better way is using the Meshviewer or something like this, by adding a "write email to owner" field with a send button, using the owner informations from the node. This should be only possible from inside the freifunk network.

jplitza, 2015-09-19:
My predecessors already said everything there is to say. Chatting is out of scope for a mesh router, and there are far better alternatives (like the link-local chat in the ChatSecure app that uses [serverless XMPP](https://xmpp.org/extensions/xep-0174.html)).

Furthermore, we would get all kinds of problems with that feature we didn't have to care about so far (validation of user input, XSS, legal problems with illegal chat messages - especially when nodes have public IPv6 addresses).

closed 2015-09-19 by jplitza