freifunk-gluon/gluon #346: Thee contacts for each node

rubo77, 2015-04-25:
To minimize the chance of nodes, that are lost and no contact is reacheable, we should add two more contact fields to the config mode.

As a workaround, you could change the text to tell the user to enter three komma separated contacts, but we should encourage the user to add those contacts, to prevent to have nodes in the net without any contact.

(In italy they have three godfathers for each node)

ohrensessel, 2015-05-07:
if a node owner does not want to be reachable, even one hundred contact fields will not help.
isn't that more a problem of "education"?

jplitza, 2015-09-02:
Given that the possibility to add more than one address simply by separating them (by comma, space or whatever) is already there, this is a community thing to advocate entering multiple persons or persons at all. If a community decides to take more technical measures, they can fork the corresponding package `gluon-config-mode-contact-info`. But in Gluon, contact information will always be optional and voluntary => closed.

closed 2015-09-02 by jplitza