freifunk-gluon/gluon #348: HTTPS-Everywhere and Privacy-Badger in config mode

rubo77, 2015-04-25:
Most users think they are protected and encrypted if they use Freifunk.

I think it would be important to add a disclaimer that you have to click in the config mode like:

X "I am aware, that all data is sent unencrypted through the mesh net and I have to use tools like HTTPS-Everywhere and Privacy-Badger to achieve the desired anonymity"

ohrensessel, 2015-05-07:
only node owners would see this disclaimer, therefore useless. freifunk is also about teaching people about the net and the risks, which are in no way special for freifunk. end-to-end encryption is a good solution for internet users in general.

rubo77, 2015-05-11:
I know, but it would be helpful to adjust some workflow to encourage the node owners to spread this knowledge somehow. Also to get rid of the accuse that Freifunk is not secure, which it isn't, like every other open network and therefore there's no base for the accuse.

Let's discuss here how we could achieve that.

tcatm, 2015-05-11:
This is completely unrelated to Gluon. If such a feature is desired, communities may add information to their site.conf's reboot message.

closed 2015-05-11 by tcatm