freifunk-gluon/gluon #349: WDR4900 eth0 MAC address is broken

NeoRaider, 2015-04-25:


christian1980nrw, 2015-04-26:
Openwrt patch:
https://lists.openwrt.org/pipermail/openwrt-devel/2014-October/028906.html

...to bei integrated into gluon...
...

NeoRaider, 2015-04-26:
Nope, that patch is already in Gluon.

Gluon breaks because it deletes the interfaces and re-creates them itself.

(Well, actually the OpenWrt MAC configuration is quite hacky and should be moved to the kernel...)

christian1980nrw, 2015-04-26:
When is it re-creating the interface?
Could this re-creating-process be patched like this openwrt fix?

NeoRaider, 2015-04-26:
Well, Gluon rewrites pretty much all of the network config. As a quick workaround to avoid further delaying the Gluon 2015.1 release, we could add such a hack to Gluon, but in the long run I'd like to fix this in the kernel, so the network config hack in OpenWrt can be dropped.

christian1980nrw, 2015-04-26:
It would be great if you could release a quick patch first, to prevent issues in the mesh nework.
I agree that a kernel fix in the openwrt (and gluon) kernel would be better.

NeoRaider, 2015-04-27:
I've ordered a test device. Once I've got the hardware, fixing this in the kernel should be straightforward.

christian1980nrw, 2015-04-28:
Good decision.
https://forum.freifunk.net/t/test-der-fastd-leistung/2368

NeoRaider, 2015-04-30:
Fixed by 531520032cdc9ab8b22b9c56d506fc9e9dd5c012

closed 2015-04-30 by NeoRaider

christian1980nrw, 2015-04-30:
I´ll try it.
Please commit this to the openwrt community too for better device support.
Thank you.