freifunk-gluon/gluon #358: Compiling problems

christian1980nrw, 2015-04-30:
Hello, i am confirming this issue with newest sourcecode.

https://forum.freifunk.net/t/gesammelte-probleme-beim-firmwarebau/4764

Collected errors:
 * opkg_install_cmd: Cannot install package gluon-config-mode-autoupdater.
 * opkg_install_cmd: Cannot install package gluon-config-mode-hostname.
 * opkg_install_cmd: Cannot install package gluon-config-mode-mesh-vpn.
 * opkg_install_cmd: Cannot install package gluon-config-mode-geo-location.
 * opkg_install_cmd: Cannot install package gluon-config-mode-contact-info.
 * opkg_install_cmd: Cannot install package gluon-luci-private-wifi.
 * opkg_install_cmd: Cannot install package gluon-luci-admin.
 * opkg_install_cmd: Cannot install package gluon-luci-autoupdater.
 * opkg_install_cmd: Cannot install package gluon-luci-portconfig.
make[4]: *** [package_install] Error 255
make[4]: Leaving directory `/freifunk/gluon/build/mpc85xx-generic/openwrt'
make[3]: *** [image] Error 2
make[3]: Leaving directory `/freifunk/gluon/build/mpc85xx-generic/openwrt'
make[2]: *** [image/TLWDR4900] Error 2
make[2]: Leaving directory `/freifunk/gluon/build/mpc85xx-generic/openwrt'
make[1]: *** [call_image/TLWDR4900] Error 2
make[1]: Leaving directory `/freifunk/gluon/build/mpc85xx-generic/openwrt'
make: *** [all] Error 2

NeoRaider, 2015-04-30:
Your site.mk isn't updated for the latest master. See http://gluon.readthedocs.org/en/latest/releases/v2015.1.html#site-changes and my mail on the Gluon mailing list last week.

closed 2015-04-30 by NeoRaider

christian1980nrw, 2015-04-30:
Another new prob:
mkdir -p /freifunk/gluon/build/mpc85xx-generic/openwrt/dl
/freifunk/gluon/build/mpc85xx-generic/openwrt/scripts/download.pl "/freifunk/gluon/build/mpc85xx-generic/openwrt/dl" "firmware-2-ct-full-community-13.bin" "a3decc529b0171ae9ec4495089ab354a" "http://www.candelatech.com/downloads/"
--2015-04-30 06:19:52--  http://www.candelatech.com/downloads/firmware-2-ct-full-community-13.bin
Connecting to 192.0.1.103:8080... connected.
Proxy request sent, awaiting response... 502 Host not found
2015-04-30 06:19:57 ERROR 502: Host not found.

Download failed.
--2015-04-30 06:19:57--  http://mirror2.openwrt.org/sources/firmware-2-ct-full-community-13.bin
Connecting to 192.0.1.103:8080... connected.
Proxy request sent, awaiting response... 404 Not Found
2015-04-30 06:19:57 ERROR 404: Not Found.

Download failed.
--2015-04-30 06:19:57--  http://downloads.openwrt.org/sources/firmware-2-ct-full-community-13.bin
Connecting to 192.0.1.103:8080... connected.
Proxy request sent, awaiting response... 404 Not Found
2015-04-30 06:19:57 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[4]: *** [/freifunk/gluon/build/mpc85xx-generic/openwrt/dl/firmware-2-ct-full-community-13.bin] Error 2

NeoRaider, 2015-04-30:
Meh. `www.candelatech.com` seems to have DNS issues at the moment, you can download the file from http://home.universe-factory.net/neoraider/firmware-2-ct-full-community-13.bin and manually put it to `openwrt/dl/`.