freifunk-gluon/gluon #372: Configurable announce interval per site.conf

Rugosh, 2015-05-13:
It would be great if the announce interval and the announced information (nodeinfo, statistics, neighbors) could be configured within the site.conf.
This feature is from my point necessary to reduce the traffic of not used information. In combination with mobile routers the basic traffic is to high. If this traffic could be reduced by disabling the statistics and neighbors information (by site.conf) that would be great.

NeoRaider, 2015-05-13:
The announce interval of alfred is hardcoded in the alfred code (we already patched that, so the information is sent only once a minute instead of every 10 seconds).

In Gluon 2015.2, we'll drop alfred support and only use the facilities provided by gluon-announced, which use a request-response scheme instead of periodical announcements (and requests can be sent via multicast).

Rugosh, 2015-05-13:
Thank you for the fast response. So if 2015.2 is released this "basic traffic" will not be routed over the whole network :smile:. In first place this sounds great but in the second this is extra work on the map server to get poll the data.

NeoRaider, 2015-05-13:
Currently, it isn't flooded over the network either. The data is unicasted to the "best" alfred master, discovered by alfred's master announcements.

closed 2015-07-12 by NeoRaider