freifunk-gluon/gluon #379: gluon-neighbour-info in 2015.1 still uses wlan0

paulinsche, 2015-05-17:
I think, that gluon-neighbour-info still uses wlan0, even with mesh_vlan enabled. Shouldn't mesh0 or mesh0.<vlan> be used?

NeoRaider, 2015-05-17:
The only script that uses gluon-neighbour-info in Gluon 2015.1 is `/lib/gluon/status-page/www/cgi-bin/status` from the `gluon-status-page` package. This script iterates over all IBSS interfaces and will thus correctly find `mesh0`, etc.

You are right though that this is broken with `mesh_vlan`, as it should use `mesh0.X`... we should just use the output of `batctl if` instead to get the list of interface to call `gluon-neighbour-info` and cut off the VLAN suffix for the `iw station dump` call. Then the only thing left would be to filter out non-WLAN interfaces.

paulinsche, 2015-05-18:
Shouldn't neighbours found also on LAN interfaces? Mesh-on-WAN was possible and Mesh-on-LAN was just introduced in 2015.1.

tcatm, 2015-05-18:
If anyone is patching this, it should be ported to the [status-page-api branch](https://github.com/freifunk-gluon/gluon/tree/status-page-api). I'm not sure whether we could support this in a sane way (e.g. without exposing VLAN details to the Javascript frontend).

NeoRaider, 2015-10-27:
The new status page has been merged, so this is now obsolete.

closed 2015-10-27 by NeoRaider