freifunk-gluon/gluon #382: Request: Default installation of siproxd to fix sip over mesh

christian1980nrw, 2015-05-20:
I was not able to receive incoming voip calls (Client: Fritz Box) over the mesh nework untill i installed siproxd on my gluon router. Siproxd is working transparent.

For Example the wdr4900 needs this package:
https://downloads.openwrt.org/barrier_breaker/14.07/mpc85xx/generic/packages/telephony/siproxd_0.8.1-4_mpc85xx.ipk

I am still testing, but i think incoming calls are working now with this config:
/etc/config/siproxd

config siproxd general
	option if_inbound		bat0
	option if_outbound		br-client

Could siproxd be integrated into gluon by default?

tcatm, 2015-05-20:
Gluon comes with no pakages integrated by default. It's all up to the community's site.mk where packages can be selected.

Can you explain what your setup does and how it fixes the problem? What is the problem? In most Gluon networks it should just work and I've been using SIP successfully over the mesh a few times.

christian1980nrw, 2015-05-20:
I think it is a similar problem like this:
http://www.sipgateblog.de/sipgate-am-ipv6-kabelanschluss/
For example sipgate recommends to use tcp keepalives (last block of the text).
And Siproxd is uing keepalives (20 secs) by default. My old Fritz Box cannot do this.

tcatm, 2015-05-20:
That's not the only possible solution. If your Fritz Box can't do that, you should update it's firmware. siproxd may also be deployed at your community's internet gateway.

closed 2015-05-20 by tcatm

christian1980nrw, 2015-05-20:
I still think it has to be integrated to fix it. There is no disadvantage.
Mobile sip user dont like to set up keepalives for freifunk to stay reachable, especially if they dont know that there is a problem.

tcatm, 2015-05-20:
Please evaluate whether this can be solved by installing siproxd on the gateways instead.