freifunk-gluon/gluon #384: fastd tries to connect via br-client

paulinsche, 2015-05-23:
As soon as mesh-vpn is configured, fastd trys to connect to the configured peers, even, if br-wan is down. fastd should try to connect only via br-wan.

NeoRaider, 2015-05-23:
Is this about IPv4 or IPv6? Have you changed anything of the network configuration of your node?

The IPv4 case should not occur, as br-client is not supposed to have IPv4 at all on a Gluon node.

The IPv6 case can't occur either when the network setup is correct. fastd packets should have a packet mark which sends them to an alternative routing table which only contains the br-wan routing table.

paulinsche, 2015-05-23:
I did not change the config. As I wrote, I can't see any outgoing packets. Just log entries like:

Sat May 23 17:06:23 2015 daemon.info fastd[5025]: sending handshake to <mesh_vpn_backbone_peer_fanlin>[[2001:4ba0:fff1:49::2]:14242]...

As there is no IPv4 route in br-client, fastd will not connect. As there is a IPv6 route in br-client, fastd tries to connect, but, as you wrote, can't, as the packets are send to an alternative routing table. Only problem is, that we got lot of log. Still a bug?

Would be nice, if you can show me, where to find packet remarking and the alternative routing table.

NeoRaider, 2015-05-25:
The packet mark 1 is set in `/etc/config/fastd`, the rule6 `wan6_lookup` in `/etc/config/network` sents all IPv6 traffic with the 1 bit set into the table 1 (see `ip -6 route show table 1`), and the `ip6table` setting in the `wan6` section of `/etc/config/network` makes netifd put the WAN IPv6 routes into table 1.

Anyways, the log message is completely normal, it just says that fastd tries to send a handshake. If there is no IPv6 route on the WAN interface, it will fail to do so, but you won't see that in the log.

If you have a peer limit, fastd should stop trying to send handshakes as soon as enough connections are established. If you don't have WAN connectivity at all, you should not enable fastd.

I don't see any bug here, everything seems to work as expected.

closed 2015-05-25 by NeoRaider