freifunk-gluon/gluon #385: Change Ad-Hoc to 802.11s "mesh"

RubenKelevra, 2015-05-27:
We're currently testing 802.11s as replacement for the ad-hoc network.

We removed the ad-hoc network completely and added a mesh-network with the following definition:

`config wifi-iface 'mesh_radio0'`  
`        option ifname 'mesh0'`  
`        option device 'radio0'`  
`        option mode 'mesh'`  
`        option network 'mesh_radio0'`  
`        option mesh_id 'wk-v15'`  
`        option mesh_ttl '1'`  
`        option mesh_fwding '0'`  

We did a 2 days test with wifi-mesh and lan-mesh, so the wlan-mesh did run idle without any issues.

After this we did a small speed and latency test, over one level we got a 0,4% packetloss with flood-ping. And around 6/1 Mbit to the Internet - this values are comparable to an ad-hoc network at this location.

We plan to setup a small mesh-network with 5-10 nodes to test over some weeks with higher load.

We going to report here over our results. :)

We really like to have a switch in the site.conf to switch between ad-hoc and 802.11s, to easily deploy a new firmware with 802.11s.

tcatm, 2015-05-27:
I've tried similar setups before and my results were comparable. We definitely would welcome 11s as an IBSS substitute in gluon. It would probably require restructuring the wifi sections in `site.conf` to allow both ad-hoc and IBSS.

Would you be willing to write or extend a Gluon package to support this? I've got this on my todo myself but I may not have time for it soon. This may be a starting point: https://github.com/freifunk-gluon/packages/pull/105

RubenKelevra, 2015-05-27:
Sure, if we completed the testings and are willing to change I would implement it for us anyways. So I gonna share it :)

RubenKelevra, 2015-06-07:
I started to implement 802.11s :)

https://github.com/VfN-NRW/gluon/compare/beta...VfN-NRW:implement_802.11s

RubenKelevra, 2015-07-12:
We have two test-installations, and it seems that an 802.11s-Mesh is about 5 % faster than an Ad-Hoc-Mesh.

closed 2015-08-08 by RubenKelevra

RubenKelevra, 2015-08-08:
@tcatm @NeoRaider  Since you decided to implement 802.11s for your own, ignoring my work without any notice, I had closed this ticket.