freifunk-gluon/gluon #386: 5GHz channel property ignored for AP (client1 interface)

scosu, 2015-05-30:
I noticed that the 5GHz AP interface is on channel 36 although the site.conf has it set to channel 40, HT40+. This breaks the mesh interface on 5ghz as it tries to join the mesh on channel 40.

Disabling the client1 interface in uci leads to a working mesh1 interface which then shows up on channel 40 correctly.

I noticed this while debugging the mesh1 interface in gluon2015.1. But it seems this problem already existed in 2014.* as all my other 5GHz router operate on the wrong channel as well.

This is the site.conf I used:
https://github.com/ffhi/ffhi-site/blob/7d72e9a8d2d87f3d7c31bc1be02f587363d86ab7/site.conf

NeoRaider, 2015-06-05:
I can't reproduce this on a WDR4900, neither with 2015.1 nor with the current (CC-based) master. I'll check with a WDR3600 next week. I've used your site.conf to build a firmware myself, if you provide your firmware images, I can test those as well.

scosu, 2015-06-06:
Unfortunately I don't have any other 5ghz routers around to test this. I further tested this with different 5ghz channels:

channel 36:

        Interface client1
                ifindex 17
                wdev 0x100000004
                addr c2:4c:02:40:62:d4
                ssid freifunk-hi.de_5ghz
                type AP
                wiphy 1
                channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz

        mesh1 interface active

channel 40:

        Interface client1
                ifindex 17
                wdev 0x100000004
                addr c2:4c:02:40:62:d4
                ssid freifunk-hi.de_5ghz
                type AP
                wiphy 1
                channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz

        mesh1 interface inactive

channel 44:

        Interface client1
                ifindex 17
                wdev 0x100000004
                addr c2:4c:02:40:62:d4
                ssid freifunk-hi.de_5ghz
                type AP
                wiphy 1
                channel 48 (5240 MHz), width: 40 MHz, center1: 5230 MHz

        mesh1 interface inactive

channel 48:

        Interface client1
                ifindex 17
                wdev 0x100000004
                addr c2:4c:02:40:62:d4
                ssid freifunk-hi.de_5ghz
                type AP
                wiphy 1
                channel 48 (5240 MHz), width: 40 MHz, center1: 5230 MHz

        mesh1 interface active

scosu, 2015-06-06:
The images I built are here. They are configured to work on channel 36 right now, but changing the channel to 40 or 44 shows this issue again.

http://firmware.freifunk-hi.de/experimental/factory/gluon-ffhi-0.7.0-tp-link-tl-wdr3600-v1.bin

NeoRaider, 2015-07-01:
Okay, seems like OpenWrt switches to the channel of other devices using the same cell ID on adhoc, so if there's another node on channel 36, it will switch to the same channel under certain circumstances.

I'll try to find out how to circumvent this issue.

NeoRaider, 2015-07-06:
I think the basic issue is that hostapd sometimes decides to switch the primary and secondary channel when HT40 is used, as hostapd only managed the AP network and doesn't know it breaks adhoc doing this. As a workaround, HT20 can be used. I think this issue might affect 2.4GHz as well.

Maybe there's a way to prevent this in hostapd...

NeoRaider, 2015-07-06:
I've looked at the hostapd code now... my conclusion:

* This affects 5GHz only
* Setting HT20 will prevent this issue

scosu, 2015-07-08:
Thanks for debugging this issue. We will use HT20 then for the moment.

I am not sure how it is coded right now. But perhaps hostapd has proper error handling so first the adhoc network could be setup and then the AP interface? Wouldn't this prevent the AP interface to switch to a different channel?

NeoRaider, 2015-07-08:
hostapd doesn't know anything about the adhoc interface, it handles only the AP interface. In my tests, the adhoc interface lost its configuration (was shown as NO-CARRIER) when hostapd decided to switch the channel...

NeoRaider, 2015-07-27:
This should be fixed by 75cebd236db7674eaea64da60ede5c8d2049661c, please test.

closed 2015-08-06 by NeoRaider