freifunk-gluon/gluon #392: Package: kmod-pppoe kernel panics under gluon but not openwrt

1977er, 2015-06-09:
Installing the package "kmod-pppoe" sends the router into a boot loop on startup. The kernel panics.
I cannot reproduce this on a vanilla OpenWRT BB.

```
U-Boot 1.1.4 (Jun 17 2013 - 12:31:57)

U-boot DB120

DRAM:  128 MB
id read 0x100000ff
flash size 8MB, sector count = 128
Flash:  8 MB
Using default environment

PCIe Reset OK!!!!!!
In:    serial
Out:   serial
Err:   serial
Net:   ag934x_enet_initialize...
No valid address in Flash. Using fixed address
 wasp  reset mask:c03300 
WASP  ----> S17 PHY *
: cfg1 0x7 cfg2 0x7114
eth0: ba:be:fa:ce:08:41
athrs17_reg_init: complete
eth0 up
eth0
Autobooting in 1 seconds
## Booting image at 9f020000 ...
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 3.10.49 (gluon@gluon) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r44952) ) #1 Tue May 19 12:43:34 CEST 2015
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9344 rev 2
[    0.000000] Clocks: CPU:560.000MHz, DDR:450.000MHz, AHB:225.000MHz, Ref:40.000MHz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line:  board=TL-WDR4300 console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 126052k/131072k available (2355k kernel code, 5020k reserved, 619k data, 288k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Calibrating delay loop... 278.93 BogoMIPS (lpj=1394688)
[    0.070000] pid_max: default: 32768 minimum: 301
[    0.070000] Mount-cache hash table entries: 512
[    0.080000] NET: Registered protocol family 16
[    0.080000] MIPS: machine is TP-LINK TL-WDR3600/4300/4310
[    0.090000] registering PCI controller with io_map_base unset
[    0.310000] bio: create slab <bio-0> at 0
[    0.310000] PCI host bridge to bus 0000:00
[    0.320000] pci_bus 0000:00: root bus resource [mem 0x10000000-0x13ffffff]
[    0.320000] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.330000] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.330000] pci 0000:00:00.0: invalid calibration data
[... 118 lines cut ...]
[   23.240000] Lo    : 00000000
[   23.240000] epc   : 872562e0 0x872562e0
[   23.240000]     Not tainted
[   23.240000] ra    : 872562d0 0x872562d0
[   23.240000] Status: 1100d403	KERNEL EXL IE 
[   23.240000] Cause : 00800010
[   23.240000] BadVA : 00000007
[   23.240000] PrId  : 0001974c (MIPS 74Kc)
[   23.240000] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv4 mac80211 ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_nat_irc nf_nat_ftp nf_nat nf_defrag_ipv4 nf_conntrack_irc nf_conntrack_ftp macvlan iptable_raw iptable_mangle iptable_filter ipt_REJECT ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 crc_ccitt compat sch_teql sch_sfq sch_red sch_prio sch_netem sch_htb sch_gred sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc batman_adv libcrc32c crc16 ip6t_REJECT ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables nf_conntrack_ipv6 nf_conntrack nf_defrag_ipv6 tun ipv6 arc4 crypto_blkcipher gpio_button_hotplug act_police cls_basic sch_tbf sch_ingress crc32c crypto_hash
[   23.240000] Process netifd (pid: 998, threadinfo=8706a000, task=87913240, tls=77d51750)
[   23.240000] Stack : 00000000 00000000 fffffff4 86dfd000 ffffffeb 00000000 00000008 86dfd000
	  80338bac 00000000 00000000 80339db0 7fc91ca0 801c8524 00000000 8706be78
	  00000000 ffffffec 86dfd000 00000000 8706be38 00000000 7f97e1e8 802032a4
	  00000000 86d65f04 86dfeb1c 8706bf10 00000000 86dfd000 00000000 80105c58
	  00000024 86d79c80 8706be78 7f97e1da 8706be88 00008924 86dfd000 80103d80
	  ...
[   23.240000] Call Trace:
[   23.240000] [<872562e0>] 0x872562e0
[   23.240000] 
[   23.240000] 
Code: 24170001  241e0040  02b2b021 <8ed00000> 12000006  00000000  8e020190  10530015  26110030 
[   23.490000] ---[ end trace f7d16be1d5e8cb4f ]---
[   23.500000] Kernel panic - not syncing: Fatal exception in interrupt
[   23.500000] Rebooting in 3 seconds..
```

NeoRaider, 2015-06-09:
Is this Gluon 2014.4 or 2015.1?

1977er, 2015-06-09:
2015.1 ... the local brew for Hannover to be exact: FFH 0.10i compiled with these params: http://hannover.freifunk.net/download/site-ffh/

NeoRaider, 2015-06-16:
This should now be fixed in the v2015.1.x branch. A maintenance release v2015.1.1 will be released very soon.

closed 2015-06-16 by NeoRaider

1977er, 2015-06-17:
Are there any details I can learn from? ;-)

NeoRaider, 2015-06-17:
http://git.openwrt.org/?p=14.07/openwrt.git;a=commit;h=64ae631f20eb349b47dae30c461ab33b5c4ac5c2 removed the kernel patch http://git.openwrt.org/?p=14.07/openwrt.git;a=blob;f=target/linux/generic/patches-3.10/617-netfilter_skip_filter_sysctl.patch;h=a570834dc665200ae03485e00049130070d87d78;hb=873779207228d20b70f9ad9aaeb844b3df17f84e , which changed the kernel ABI (the size of `struct netns_ct` was changed). I've replaced it with https://github.com/freifunk-gluon/gluon/blob/v2015.1.x/patches/openwrt/0039-Revert-netns_ct-ABI-change-introduced-in-r44874.patch to restore ABI compatiblity.

1977er, 2015-06-17:
Ah. Thanks. :)