freifunk-gluon/gluon #397: mesh-vpn (fastd) broken in IPv6 only configuration

benerudolph, 2015-06-16:
This was tested on the Freifunk Darmstadt Infrastructure using our build of gluon 2015.1.
Config-mode supports enabling and disabling the wan and wan6 interfaces. Disabling IPv4 (setting 'none') leads to protocol 'none' instead of 'dhcp' in /etc/config/network.
IPv6 apparently works but fastd is unable to connect to any of the four gateway servers.

This configuration is known to be working on gluon 2014.4 on a Unitymedia DS-Lite (CG NAT) home network.

dracoTrier, 2015-06-16:
'IPv6 only' works for us when set up using ssh + prepending "ipv6 " to the gateway addresses (fastd.mesh_vpn_backbone_peer_Gatewayname.remote). Could you try if this also works for you?

NeoRaider, 2015-06-17:
Please provide:
* `/etc/config/network`
* site.conf
* Output of `ip a` and `ip r s t all` in the broken state

benerudolph, 2015-06-17:
Here is the `/etc/config/network` of the affected node 
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd83:786d:fc44::/48'

config interface 'wan'
	option enabled '1'
	option igmp_snooping '0'
	option ifname 'eth0'
	option auto '1'
	option peerdns '0'
	option type 'bridge'
	option proto 'none'

config interface 'wan6'
	option enabled '1'
	option ifname 'br-wan'
	option ip6table '1'
	option peerdns '0'
	option proto 'dhcpv6'

config rule6 'wan6_lookup'
	option mark '0x01/0x01'
	option lookup '1'

config route6 'wan6_unreachable'
	option type 'unreachable'
	option table '1'
	option target '::/0'
	option metric '65535'
	option gateway '::'
	option interface 'loopback'

config interface 'client'
	option reqprefix 'no'
	option ifname 'bat0'
	option proto 'dhcpv6'
	option type 'bridge'
	option igmp_snooping '0'
	option macaddr 'e8:de:27:74:87:58'
	option peerdns '1'

config interface 'mesh_wan'
	option auto '0'
	option ifname 'br-wan'
	option mesh 'bat0'
	option proto 'batadv'

config interface 'mesh_vpn'
	option ifname 'mesh-vpn'
	option mesh_no_rebroadcast '1'
	option macaddr 'ea:e2:27:74:87:58'
	option mesh 'bat0'
	option proto 'batadv'

config interface 'bat0'
	option ifname 'bat0'
	option macaddr 'e8:de:27:74:87:58'
	option proto 'none'

config interface 'mesh_radio0'
	option mesh 'bat0'
	option proto 'batadv'
	option mtu '1532'
[... 84 lines cut ...]
default from :: via fe80::44dd:cdff:fe55:75ed dev br-client  metric 2048 
default from 2a03:2260:118::/64 via fe80::44dd:cdff:fe55:75ed dev br-client  metric 2048 
default from fdca:ffee:ffda::/64 via fe80::44dd:cdff:fe55:75ed dev br-client  metric 2048 
2a03:2260:118::/64 dev br-client  metric 256 
unreachable fd83:786d:fc44::/48 dev lo  metric 2147483647  error -128
fdca:ffee:ffda::ffff dev local-node  metric 256 
fdca:ffee:ffda::/64 dev br-client  metric 256 
fdca:ffee:ffda::/64 dev br-client  metric 1024 
fe80::/64 dev mesh-vpn  metric 256 
fe80::/64 dev bat0  metric 256 
fe80::/64 dev br-wan  metric 256 
fe80::/64 dev br-client  metric 256 
fe80::/64 dev local-node  metric 256 
fe80::/64 dev client0  metric 256 
fe80::/64 dev mesh0  metric 256 
unreachable default dev lo  metric -1  error -128
ff00::/8 dev local-node  metric 256 
ff00::/8 dev mesh-vpn  metric 256 
ff00::/8 dev bat0  metric 256 
ff00::/8 dev br-wan  metric 256 
ff00::/8 dev br-client  metric 256 
ff00::/8 dev client0  metric 256 
ff00::/8 dev mesh0  metric 256 
unreachable default dev lo  metric -1  error -128
```

NeoRaider, 2015-06-17:
Mmh, I forgot the simple busybox `ip r` doesn't display table numbers in `ip r s t all`, so that isn't as helpful as I had hoped...

What does `logread` say when fastd is trying to connect?

benerudolph, 2015-06-18:
fastd says `resolving Gatway`and than fails upon establishing the connection. I will provide a log transcript in a few more minutes. I suspect it is a DNS issue / rouing issue, because IPv6 ping apparently works.

NeoRaider, 2015-07-12:
I've been able to reproduce this issue, I'll have a closer look at it soon.

flokli, 2015-07-16:
:thumbsup:

commit d9df1186 (2015-08-06): gluon-wan-dnsmasq: fix usage of libpacketmark

Using service_start requires exporting of environment variables.

Fixes #397
PR #431 (2015-08-06): gluon-wan-dnsmasq: fix usage of libpacketmark
Using service_start requires exporting of environment variables.

Fixes #397
commit 75b3c807 (2015-08-06): gluon-wan-dnsmasq: fix usage of libpacketmark

Using service_start requires exporting of environment variables.

Fixes #397
closed 2015-08-06 by NeoRaider