freifunk-gluon/gluon #403: compressed gluon-announced data cannot be inflated

RiBro, 2015-06-23:
Hi I tried to extract statistics data from my gluon node using

     echo -n "GET statistics" | socat - UDP6-DATAGRAM:[IPv6]:1001 | zcat

and it does not work. Also other standard decompression tools do not work. Instead of zcat i tried

    gunzip
    zlib-flate -uncompress 
    python -c "import zlib,sys;print repr(zlib.decompress(sys.stdin.read()))" 

and some more standard tools. 
Ether provide uncompressed data or a tool for inflation of these

NeoRaider, 2015-06-23:
The lua-deflate module uses the raw deflated format without zlib header. It can be decompressed in python by supplying a negated "wbits" value to zlib.decompress, e.g.:

    print(repr(zlib.decompress(sys.stdin.buffer.read(), -15))) # Python3

    print(repr(zlib.decompress(sys.stdin.read(), -15))) # Python2

closed 2015-06-23 by NeoRaider

RiBro, 2015-06-23:
Thanks for the fast reply..