freifunk-gluon/gluon #404: x86-gerneric zweites Interface hängt nicht im br-client

MPW1412, 2015-06-23:
Hallo,

wie im Chat ausführlich unter Mithilfe von ranlvor ermittelt, wird das zweite Interface zwar aktiviert, aber bei deaktiviertem LAN-Mesh nicht automatisch in br-client gesteckt. Dadurch bekommen Clients keine IP und keine Verbindung.

Folgende Änderung in /etc/config/network in der Sektion 'client',

Statt:
option ifname 'bat0'

Muss es heißen:
option ifname 'bat0 eth0'

Grüße
MPW

NeoRaider, 2015-06-24:
What Gluon version is this? After initial installation, or after an upgrade?

MPW1412, 2015-06-24:
2015.1.1 fresh install, flashed with "sysupgrade -n".

NeoRaider, 2015-06-24:
Can't reproduce here. Can you upload the image you are using somewhere?

MPW1412, 2015-06-24:
https://mpw.no-ip.biz/privat/gluon-ffms-v2015.1.1+251-x86-generic.img.gz

(Dyndns, slow, self generated certificate)

NeoRaider, 2015-06-24:
Your site.conf has `mesh_on_lan` enabled.

(Also, it contains a weird `batman_on_wan` option that has never existed in a released version of Gluon).

closed 2015-06-24 by NeoRaider

MPW1412, 2015-06-24:
But still it doesn't work, if you disable mesh_on_lan:

uci set network.mesh_lan.auto=0
uci commit
reboot

Thanks for the tip with batman_on_wan, I'll remove that.

NeoRaider, 2015-06-24:
uci is a low-level configuration tool, it never does anything you don't explicitly ask it to do.

Using the `uci` command has changed the `mesh_lan.auto` setting in `/etc/config/network`, not more and not less. If you want to change `network.client.ifname` as well, you need to do that with another `uci set`.

MPW1412, 2015-06-24:
Okay, my bad, I thought toggling mesh_on_lan was as easy as mash_on_wan. Never mind, this is definitly a layer 8 problem then ;)

NeoRaider, 2015-06-24:
See https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#mesh-on-lan

MPW1412, 2015-06-24:
That's what I was looking for. Thanks!