freifunk-gluon/gluon #407: Add some aliases on the console

rubo77, 2015-07-01:
It would come handy if you had some default aliases defined on the console of the nodes like

    ll="ls -lah"
    ..="cd ../"

etc.

and most useful would be this alias that simulate something like locate with just some lines:

    alias locate="if [ ! -e /tmp/locate.db -a ! -e /tmp/locate.lockdb ]
    then touch /tmp/locate.lockdb
    trap \"rm /tmp/locate.lockdb; rm /tmp/locate.db; exit\" SIGHUP SIGINT SIGTERM
    find /|tee /tmp/locate.db
    chmod 666 /tmp/locate.db
    rm /tmp/locate.lockdb
    elif [ -e /tmp/locate.lockdb ]
    then find /
    else cat /tmp/locate.db
    fi|grep "
And an alias to refresh the locate cache:

    alias relocate="if [ ! -e /tmp/locate.lockdb ]
    then rm /tmp/locate.db
    fi
    locate dummy_for_grep"

source:
http://unix.stackexchange.com/a/212057/20661

NeoRaider, 2015-07-01:
If you like these aliases, add them yourself. Such aliases are to much a matter of taste to enable them by default.

closed 2015-07-01 by NeoRaider

rubo77, 2015-07-01:
There are already some aliases defined in `/etc/profile`:

    [ -x /bin/more ] || alias more=less
    [ -x /usr/bin/vim ] && alias vi=vim || alias vim=vi