freifunk-gluon/gluon #410: CPE wireless signal quality indicator LEDs

MichaelSasser, 2015-07-08:
The CPE210/CPE510 has some wireless signal quality indicator LEDs, displaying the received wireless signal strength of the root AP. Since the update of my CPE210 to v2015.1.x, these LEDs has not been displaying the received wireless signal strength anymore. Yes, I know, the LEDs are not rocket solid, but help a lot aligning a new installed node. It's bizarre with the node in one hand and an eye on the notebook in the other hand. 

The Problem seems to be that the ifname was changed and the leds are still using wlan0 for rssi. You will find them in /etc/config/system.

```
config rssid 'rssid_wlan0'
	option dev 'wlan0'
	option refresh '40000'
	option threshold '1'

config led 'led_rssilow'
	option name 'RSSILOW'
	option sysfs 'tp-link:green:link1'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '1'
	option maxq '100'
	option offset '0'
	option factor '13'

config led 'led_rssimediumlow'
	option name 'RSSIMEDIUMLOW'
	option sysfs 'tp-link:green:link2'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '26'
	option maxq '100'
	option offset '-25'
	option factor '13'

config led 'led_rssimediumhigh'
	option name 'RSSIMEDIUMHIGH'
	option sysfs 'tp-link:green:link3'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '51'
	option maxq '100'
	option offset '-50'
	option factor '13'

config led 'led_rssihigh'
	option name 'RSSIHIGH'
	option sysfs 'tp-link:green:link4'
	option trigger 'rssi'
	option iface 'rssid_wlan0'
	option minq '76'
	option maxq '100'
	option offset '-75'
	option factor '13'
```

NeoRaider, 2015-07-10:
Should be fixed by f5764a4a7b68f2eb56c9c5f2995a6277bf1b4aaa, please test. If it is working, I'll also backport the fix to the v2015.1.x branch.

MichaelSasser, 2015-07-12:
Nice! I'm now building a new Version on branch v2015.1.x. I had a couple of issues after a "dirclean". The Openssl version 1.0.2c is not available anymore, so I manually updated openssl to version [1.0.2d](https://mta.openssl.org/pipermail/openssl-announce/2015-July/000037.html) (current stable). There are also a couple of changes needed in the patchfiles. I think that I have done almost the same like the openwrt guys in [here](https://github.com/openwrt-mirror/openwrt/commit/824beb9a7f1e0c2943e4f6da6af1586b71b85aed). I will test the new firmware later that day.

NeoRaider, 2015-07-12:
The fix is currentlly only in the Gluon master, not in v2015.1.x.

MichaelSasser, 2015-07-12:
I've created the "350-gluon-mesh-batman-adv-core-rssid" in v2015.1.x with the same content. This should work, shouldn't it? On a simple sysupgrade without wiping the config the option is still wlan0. I will try it on master branch.

closed 2015-07-12 by MichaelSasser