freifunk-gluon/gluon #413: Raspberrypi: Setting password via config mode fails

MPW1412, 2015-07-12:
Hello,

it's not possible to set a password via the webinterface in configuration mode. (Uploading a public key works.)

    /usr/lib/lua/luci/dispatcher.lua:364: Failed to execute cbi dispatcher target for entry '/admin/remote'.
    The called action terminated with an exception:
    /usr/lib/lua/luci/sys.lua:460: attempt to concatenate local 't' (a nil value)
    stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:364: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:135: in function </usr/lib/lua/luci/dispatcher.lua:134>

NeoRaider, 2015-07-12:
Thanks for the report. Not RasPi-specific, affects all targets in master.

NeoRaider, 2015-07-12:
Fixed.

closed 2015-07-12 by NeoRaider

MPW1412, 2015-07-12:
I already wondered, why this happend especially on the Pi. But as I didn't test master recentely, I hadn't noticed it on other targets.

Thanks for the quick fix!