freifunk-gluon/gluon #414: gluon-config-mode: make hostname field a "pretty hostname"

NeoRaider, 2015-07-12:
A "pretty hostname", as it is used by many desktop systems, may be any unicode string and can be used as a human-readable node identifier. The actual "technical" hostname is derived from the pretty hostname to be compliant with RFC1123.

* We should have a look at the algorithms other systems use for the derivation of the actual hostname (e.g. systemd's hostnamectl)
* Do we need a minimum number of "valid" characters in the pretty hostname, or do we use a fallback hostname when the pretty hostname contains no ASCII-mappable characters?

rubo77, 2015-12-04:
Current status: This line:
`o.datatype = "hostname"`
in
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-hostname/files/lib/gluon/config-mode/wizard/0100-hostname.lua#L12
is causing that the `hostname` is validated by openwrt in:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/cbi/datatypes.lua#L180

Couldn't we change that line so it is not of type `hostname`, but an own datadype, that we can add our own validation?

Preferably that validation would be configurable in site.conf, so you can have your own specialties in each community

NeoRaider, 2015-12-04:
Changing the datatype would be one part of implementing this, but obviously, we also need to change the backend part. We need to store the pretty hostname somewhere, generate the real hostname from it (preferably the same way other OSs are doing this), and then store the real hostname to the `system` UCI package.

I don't see what should be configurable in the site.conf, the hostname should be an arbitrary identifier. If communities want other metainformation about the nodes, they should add additional fields and not abuse the hostname field...

rubo77, 2015-12-04:
Each community could have a certain rule how the hostname should look like in their community, for example:

 * must start with <PLZ>-<TOWN>-
 * must start with your subcommunity-identifyer
 * may not contain "freifunk"
 * may not start with "ff-"
 * ...

NeoRaider, 2015-12-04:
@rubo77, I don't think that makes sense. The "pretty hostname" can be any non-empty string. The "actual hostname" it the same as the pretty hostname, but cleaned of everything not compliant with RFC1123.

Things like PLZ or a subcommunity-identifier don't belong in the hostname. Adding additional fields to the config mode and the announced data for this would be easy and can be done in community-specific packages.

rubo77, 2015-12-04:
I think it is useful to add information in the hostname itself, example: http://www.freifunk-aachen.de/map/v2/index.html

also you are right, you could **add** extra information in an extra field, but you could not **prohibit** or **remove** certain characters/words, that would **not be allowed** in the hostname.

For this you would have to modify this host-input procedure itself (which is the main issue here)

NeoRaider, 2015-12-04:
I think any character should be allowed in the pretty hostname. It may be *any* string.

rubo77, 2015-12-04:
Yes, that is true. I could redefine this in a separate issue but I only thought that we could kill two birds with one stone here in this issue and add some extra functionality to add a blocklist or rule for the hostname here too.

closed 2016-08-27 by NeoRaider

rubo77, 2016-11-28:
I think in some communities it is not desired to allow all characters.

can we make the `pretty_hostname` optional?

rotanid, 2016-11-28:
i'm pretty sure no one would stop you from making a PR to achieve this.
but in your use case, you would better add a check that throws an error if the default hostname remains unchanged.

NeoRaider, 2016-11-28:
I don't see a reason to disallow specific characters. All software processing respondd/alfred data must be able to handle any kind of weird input anyways (or at least gently fail), as nobody guarantees that all received data is sent by our code at all.

Using invalid content was a really ugly hack, as this even set the invalid string as the kernel hostname during the config mode. We certainly won't support such hacks in any way, please find a saner solution.

rubo77, 2016-11-28:
Would this be a solution? https://github.com/freifunk-kiel/ffki-packages/commit/5adafdc5300ad298e0d027d764c0f5d7cf8a696a

I copied the package and just added the check

     o.datatype = "hostname"

to the save process, so the old behaviour.

commit 7e007558 (2016-08-27): Use a "pretty" hostname where possible

Fixes #414