freifunk-gluon/gluon #416: respondd needs too much RAM

NeoRaider, 2015-07-12:
In the current setup, respondd is the process with the highest RAM usage (as it always keeps a Lua context allocated to get high performance).

tcatm, 2015-09-01:
https://github.com/freifunk-gluon/gluon/commit/363fe96b60b94cd2cd16ce29e0908329e7f486d1 and https://github.com/freifunk-gluon/gluon/commit/fc210188bc92990c9d1ce04a2e27ada59b905ee0 reduce the heap size of respondd to ~300kb. I'll make a PR once #465 is merged.

NeoRaider, 2015-12-20:
I guess having a long-running Lua service is just not feasible on devices with 32MB RAM. I have an idea how to rewrite respondd to use C modules instead of Lua snippets, I'll prepare a pull request soon.

NeoRaider, 2016-02-02:
Fixed as well as possible in 9004028cb47d360360c9fdcc2fea52822998a75f and freifunk-gluon/packages@fd06c7d67da69713f0361dadab16393c26cb609b

closed 2016-02-02 by NeoRaider