freifunk-gluon/gluon #419: Route from WAN to Freifunk

mist, 2015-07-21:
I want my computers to have access to the Freifunk network without loosing access to the local network. 
So I want to add another Route in my FRITZ!Box-setup for ipv4 and for ipv6 my Freifunk router should advertise it's IPv6-prefix to the fritzbox (I hope I'm understanding this correctly). The fritzbox option is called "Auch IPv6-Präfixe zulassen, die andere IPv6-Router im Heimnetzwerk bekanntgeben"

tcatm, 2015-07-21:
Gluon does not support such setups out the box. You may, however, install a routing daemon like `bird` for distributing routes using an IGP within your mesh. This will require support from your community. In Lübeck, we use RIPng for this purpose. This wiki article may help you: https://wiki.luebeck.freifunk.net/FAQ/Privates-Netz-einrichten

jplitza, 2015-09-02:
As an additional note, what you had in mind wouldn't work anyway: The node could send router advertisements to your LAN containing the prefix from the mesh, but either that would make it the default gateway for IPv6 in general, which probably isn't desired, or no gateway it all. In the latter case, you would have IPv6 addresses in your LAN, but no route (direct) to the Freifunk network via your node.

For IPv4, as tcatm already said, the rest of the Freifunk network would also need a route to your home network. To achieve that, you would definitely need a routing daemon like bird, which doesn't fit in the small 4MB flash memory of most routers supported, so there won't be a global implementation. Thus, closing.

closed 2015-09-02 by jplitza