freifunk-gluon/gluon #423: Error after remote-password

lwulfe, 2015-07-26:
http://192.168.1.1/cgi-bin/luci/admin/remote on TL-WR841N Version 9.2
Sending password results in
/usr/lib/lua/luci/dispatcher.lua:364: Failed to execute cbi dispatcher target for entry '/admin/remote'.
The called action terminated with an exception:
/usr/lib/lua/luci/sys.lua:460: attempt to concatenate local 't' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:364: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:135: in function </usr/lib/lua/luci/dispatcher.lua:134>

Gluonversion: gluon-v2015.1-21-g449a18f

tcatm, 2015-07-26:
It looks like that Gluonversion has local changes on top of an old release. Can you try again using the latest master? What steps are necessary to reproduce this? Setting the password worked fine for me yesterday.

NeoRaider, 2015-07-26:
You are using an old master version (commit 449a18f is from June 15th), this bug has been fixed already (in commit e8781556011a94c3f6deea5063e80110d450eae3)

closed 2015-07-26 by NeoRaider