freifunk-gluon/gluon #425: Positionierung zur Gluon Firmware

A-Kasper, 2015-07-27:
Heute hat einer der Hauptentwickler (tcatm) das Repository des Meshviewers gelöscht. 
Siehe: https://forum.ffrl.de/t/ffmap-backend-und-meshviewer/4123/226

Er hat das Projekt damit zu seinem eigenen gemacht und verhindert aktiv eine offene Entwicklung und unabhängige Forks und Branches. In meiner Wahrnehmung ist der Freifunk ein Community Projekt. Alternative Entwicklungen sollten möglich sein. Insbesondere sollte es möglich sein das Projekt auch dann weiterzuentwickeln, wenn die zur Zeit Hauptaktiven nicht mehr zur Verfügung stehen. 

Ich bitte in Sachen gluon um eine Klarstellung, ob ähnliches hier denkbar, oder gar bereits geplant ist. Ich bedauere sehr, dass einer der Hauptentwickler sich vom Community Gedanken verabschiedet hat und Mitgestaltung bzw. eigene Derivate unterbindet. Ich kann nachvollziehen, wenn der zurückfliessende Code ihm qualitativ nicht gut genug ist, allerdings muss er ihn ja auch nicht übernehmen. Zu verhindern, dass andere mit anderen Ansprüchen ihn weder nutzen noch weiterentwickeln können ist ein Schritt in die Falsche Richtung. 

Deshalb die Bitte um Klarstellung:
Müssen wir damit rechnen auch das gluon repository gelöscht und nur noch vorcompilierte Software zur Verfügung gestellt wird? Könnt ihr ausschließen, dass dies hier geschieht?

Diese Frage ist sicherlich für viele wichtig, die bisher auf gluon setzen. 

Vielen Dank.

tcatm, 2015-07-27:
Meshviewer ist ein privates Projekt von mir mit dem ich Visualisierungsmethoden für Freifunknetze erforsche.

Gluon ist ein Communityprojekt.

A-Kasper, 2015-07-27:
Das war nicht die Antwort auf die Frage:
Ich möchte eine klare Antwort dazu ob hier ein ähnliches vorgehen denkbar wäre. Immerhin hast Du gerade angekündigt eine Version ohne lua und shellscripte erstellen zu wollen...

NeoRaider, 2015-07-27:
Mit Gluon wird es so etwas definitiv nicht geben. Dieses Github-Reposory ist das Zentrum der Entwicklung, und es wird auch so bleiben. Die Community ist für die Entwiclung höchst wertvoll und liefert immer wieder Beiträge zur Weiterentwicklung.

Zum Mesh Viewer kann ich nichts sagen, dieser war von Anfang an als unfertiges Privatprojekt von @tcatm angekündigt. Es kann gut sein, dass es in Zukunft wieder ein öffentliches Git-Repo geben wird, vielleicht, wenn sich der Mesh Viewer etwas weiter entwickelt hat.

A-Kasper, 2015-07-27:
Danke.

closed 2015-07-27 by A-Kasper